Ultimaker/Cura #10850: cannot multiple objects

Carinawei97, 2021-11-16:
### Application Version

4.12.0

### Platform

macOS Monterey 12.0.1

### Printer

Ultimaker S3

### Reproduction steps

1. Import a .stl file
2. select the object and choose multiple selected

### Actual results

the screen shows 'multiplying and placing objects' but the progress bar did not run, and no multiple object show in the screen
[image]

### Expected results

Multiple the selected object and place it in somewhere

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

I used different version of Ultimaker Cura (14.11 and 14.12) in my new MacBook Pro with chip M1 MAX, none of them could multiple the selected objects. But this function works well in my MacBook Air macOS Mojave 10.14.6 (1.6 GHz Intel Core i5).

ragecub3, 2021-11-17:
There is a fix for that issue maybe it will help.
https://github.com/Ultimaker/Cura/issues/10249#issuecomment-967009511

closed 2021-11-17 by nallath

Carinawei97, 2021-11-17:
> Duplicate of #10249 <https://github.com/Ultimaker/Cura/issues/10249>

Thank you very much!

It works very well now.

Siyuan