Ultimaker/Cura #10853: Duplication error

tritely, 2021-11-17:
### Application Version

4.12.0

### Platform

Mac OS Monterey, M1 Max

### Printer

Anycubic Vyper

### Reproduction steps

Importing STL file to CURA.

### Actual results

The version of Cura and previous version (4.11) has a bug of duplicating STL files while placing them on the printing plate. 

### Expected results

Thereby, I stop duplicating STL files and currently I am importing the same STL file multiple times when I need to print the same object. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

no

nallath, 2021-11-17:
Duplicate of #10852

nallath, 2021-11-17:
Whoops my bad, I marked this as the duplicate of the wrong ticket. It's marked correctly now.

closed 2021-11-17 by nallath