Ultimaker/Cura #10864: Support Line and Brim Bug

MaxThroughput, 2021-11-17:
### Application Version

4.12

### Platform

Windows 10

### Printer

Creality Ender 3 Pro

### Reproduction steps

Generate Support - Yes
Support Pattern - Lines
Support Wall Line Count - 0
Connect Support Lines - No
Enable Support Brim - No

### Actual results

Support brim is displayed in cura preview and prints accordingly. There should be no brim printed when disabled.
Line supports appear correctly disjointed in preview. However, when printing the line supports are connected like zig zags.

### Expected results

Lines should be disjointed. Support brim should not be present

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[blank.txt](https://github.com/Ultimaker/Cura/files/7556881/blank.txt)

MaxThroughput, 2021-11-17:
[image]

MaxThroughput, 2021-11-17:
Solution:

This behavior was not the result of a bug, but a misconfiguration. In the attached screenshot you can see the blue travel path differentiated from the yellow material. The connections were the result of retractions not occurring in the travel path between support lines. The support density setting is 1.14mm, while the Retraction Minimum Travel value was 1.5mm. When Retraction Minimum Travel < Support Density Line Distance, retractions occur and the result is a disconnected line support pattern.
[image]
e

fvrmr, 2021-11-23:
Hi @MaxThroughput thank you for your report. 
These lines are not the brim but the grid for the support. Good to know that you've found the solution. 
I will close this issue since your part is solved and the brim part is not a bug but expected. 
But let me know if you have more questions.

closed 2021-11-23 by fvrmr