Ultimaker/Cura #10865: Cura 4.12 -- not multiplying objects

ealmeidaus, 2021-11-18:
### Application Version

4.12

### Platform

Mac

### Printer

Creality CR-6 SE

### Reproduction steps

After upgrading to the Cura 4.12 (I had the 4.11 before) it is not possible to multiply the objects. When you select how many you want to duplicate the app just freeze.

### Actual results

Nothing happens. The progress bar remains frozen.

### Expected results

Multiply the selected object based on the qty informed.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]

fieldOfView, 2021-11-18:
See #10249

fvrmr, 2021-11-18:
Closing this issue since it is a duplicate of https://github.com/Ultimaker/Cura/issues/10249

closed 2021-11-18 by fvrmr