Ultimaker/Cura #10868: ULTIMAKER CURA 12 NO FUNCIONA

cinthialm95, 2021-11-18:
### Is your feature request related to a problem?

Hola estoy utilizando ultimaker cura y de repente dejó de funcionar... cuando le doy a segmentar se queda bloqueado y no responde.

### Describe the solution you'd like

Me podéis ayudar?

### Describe alternatives you've considered

A alguien le paso lo mismo que a mi? Como puedo solucionarlo?

### Affected users and/or printers

Creo que este problema afecta a más gente.

fieldOfView, 2021-11-18:
Please use english here. If you cannot use english, please go to the spanish subforum of the Ultimaker community forum instead. Perhaps someone there will be able to help you:
https://community.ultimaker.com/forum/125-espa%C3%B1ol/

fvrmr, 2021-11-19:
Hi @cinthialm95 thank you for your report.
Google translate tells me Cura stopped working. So if that is the case this is not a feature request. 
Please let us know in English what your problem is and share your log file: https://github.com/Ultimaker/Cura#logging-issues

closed 2021-12-03 by no-response[bot]