Ultimaker/Cura #10873: Top surface using issues

neofuture, 2021-11-18:
### Application Version

4.12.99 - Monterey Fix

### Platform

macOS M1 Montery beta 12 

### Printer

Ender 3 v2

### Reproduction steps

Printing any thing results in a poor top layer

### Actual results

Top layer seems to use concentric only, no matter what settings I change the quality of the top is no different

### Expected results

Better surface performance

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[image]
[CE3_V1.01_Chassis_Plate(RPi).3mf.zip](https://github.com/Ultimaker/Cura/files/7563040/CE3_V1.01_Chassis_Plate.RPi.3mf.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/7563043/cura.log)

fvrmr, 2021-11-18:
Hi @neofuture thank you for your report. 
We don't maintain the quality profiles for your printer. Since we can't test these, we only test Ultimaker printers. 
These profiles were contributed to us by the community. 

If you use the default Super quality for an Ender you will get lines as Top/Bottom pattern. 
Looking from the preview I can't really see if there is a bug in here.

neofuture, 2021-11-18:
OK previous to this version on version 4.11 the top/bottom surface was much better than this, I will roll back to that version and test again after this print is finished, I have not changed the profile, but the top surface seems to change if you remove the raft

neofuture, 2021-11-18:
[image]

The above is with 4.12

[image]

Above is 4.11 

Both using exactly the same profile

neofuture, 2021-11-18:
[image]

And this is 4.11 with the raft off and skirt on

neofuture, 2021-11-18:
Ok some how the profile had got something wrong with it I have removed the ender 3 profile and re added it and now its working, please feel free to close this if you think its just an issue on my end

fvrmr, 2021-11-19:
Thank you for your update! I will close this issue now since it is solved for you :)

closed 2021-11-19 by fvrmr