Ultimaker/Cura #10874: 1.st layer support omission

RudiCor, 2021-11-18:
### Application Version

4.12.0

### Platform

mint-linux

### Printer

creality CR6-MAX

### Reproduction steps

Loaded and sliced model as usual

### Actual results

region of 1.st layer without support (cross lines)

### Expected results

all support areas of 1.st layer with cross-li
[image]
ne

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CCR6MAX_1st_layer.3mf.zip](https://github.com/Ultimaker/Cura/files/7563102/CCR6MAX_1st_layer.3mf.zip)

fvrmr, 2021-11-19:
Hi @RudiCor thank you for your report.
Did you use the official AppImage of Cura?
For me the support is showing up correctly. But you project file is not sharing your settings and printer. So I could not reproduce it the right way.

Ghostkeeper, 2021-11-25:
That's not a project file. That's a plain 3MF file. A project file can be created by going to File -> Save Project...

RudiCor, 2021-11-25:
> Android [4].
> Links:

Hello,

I changed some parameters and can not reproduce the setup that produced 
the issue any more. Now it works fine.
Sorry for disturbing you!

Corneliu

closed 2021-11-26 by Ghostkeeper