Ultimaker/Cura #10875: Multiply objects does not work

Callewaert3D, 2021-11-18:
### Application Version

4.12.0

### Platform

macOS Monterey

### Printer

UltimakerS5

### Reproduction steps

I choose an object on the build plate

### Actual results

"Placing objects" window opens with progress barr "multiplying an placenta objects"

### Expected results

Object should be multiplied on the build plate

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fieldOfView, 2021-11-18:
Especially for you, Cura 4.12.1 was released 10 minutes before you posted your issue:
https://github.com/Ultimaker/Cura/releases/tag/4.12.1

Callewaert3D, 2021-11-18:
Works perfect now! 

Thanks,

P

fvrmr, 2021-11-18:
Closing this issue since it is solved!
See also https://github.com/Ultimaker/Cura/issues/10249

closed 2021-11-18 by fvrmr