Ultimaker/Cura #10876: Cura 4.12.1 file icon missing

ESPATAR, 2021-11-18:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Sermoon D1

### Reproduction steps

i just remove cura 4.12.0 then install 4.12.1 
and select .3mf .gcode .stl to default open by cura.

i uninstall 4.12.1 and reinstall but not fix.

### Actual results

[image]

file not preview 
preview only ".3MF"

### Expected results

with 4.12.0 can preview within icon

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

nothing. thanks for help

fvrmr, 2021-11-19:
Hi @ESPATAR thank you for your report.
For me it was also not working with 3mf files and I restarted my computer a couple of times to see if it would show up.
I will discuss this with a colleague. Keep you posted.

ESPATAR, 2021-11-24:
.3mf and .stl from Fusion 360 can't thumbnail. 
.3MF from solidwork can thumbnail.
.STL from solidwork can not thumbnail.

Ghostkeeper, 2021-11-25:
The STL file format has no place for a thumbnail. Whether or not you see a thumbnail is only in the operating system integrations. Cura doesn't provide those integrations in either case.

I'd expect that, if the file format is associated with an application with an icon, that Windows would show the icon of that application. It seems that it used Cura's icon for the 3MF files, but apparently not for the STL files. So clearly the icon is in the application and Windows is able to read it. But doesn't do that properly for STL?

I'm not sure we can really do anything about what Windows decides to show an icon for though.

Ghostkeeper, 2021-12-03:
@fvrmr Do you know what the outcome was of that discussion with a colleague?

fvrmr, 2022-02-01:
This is a probably a duplicate of https://github.com/Ultimaker/Cura/issues/3281 
Could you check if the workaround provided there solves your issue?

closed 2022-02-15 by no-response[bot]