Ultimaker/Cura #10877: Initial bed temperature not in G code

rsa2010, 2021-11-18:
### Application Version

4.12.0-beta

### Platform

Windows 10

### Printer

Ultimaker 2C

### Reproduction steps

Sliced a project with initial bed temperature at 100C, other layers 90C
Bed was not heated when printer started, saved the sliced result as G-Code file and checked the file.
Did not find M190 S100 in the code 

### Actual results

;START_OF_HEADER
;HEADER_VERSION:0.1
;FLAVOR:Griffin
;GENERATOR.NAME:Cura_SteamEngine
;GENERATOR.VERSION:4.12.0-BETA
;GENERATOR.BUILD_DATE:2021-10-22
;TARGET_MACHINE.NAME:Ultimaker 2+ Connect
;EXTRUDER_TRAIN.0.INITIAL_TEMPERATURE:260
;EXTRUDER_TRAIN.0.MATERIAL.VOLUME_USED:21423
;EXTRUDER_TRAIN.0.MATERIAL.GUID:a17b348e-9e68-435a-bdeb-ee933c9193d5
;EXTRUDER_TRAIN.0.NOZZLE.DIAMETER:0.4
;EXTRUDER_TRAIN.0.NOZZLE.NAME:unknown
;BUILD_PLATE.TYPE:glass
;BUILD_PLATE.INITIAL_TEMPERATURE:100
;PRINT.TIME:15742
;PRINT.GROUPS:1
;PRINT.SIZE.MIN.X:47.536
;PRINT.SIZE.MIN.Y:56.366
;PRINT.SIZE.MIN.Z:0.2
;PRINT.SIZE.MAX.X:189.117
;PRINT.SIZE.MAX.Y:163.936
;PRINT.SIZE.MAX.Z:10.55
;END_OF_HEADER
;Generated with Cura_SteamEngine 4.12.0-BETA
T0
M82 ;absolute extrusion mode

G92 E0
M109 S260
G280 S1
G0 Z20.001
G1 F1500 E-6.5
;LAYER_COUNT:70
;LAYER:0
M107
G0 F4090.9 X51.593 Y60.408 Z0.2

### Expected results

;START_OF_HEADER
;HEADER_VERSION:0.1
;FLAVOR:Griffin
;GENERATOR.NAME:Cura_SteamEngine
;GENERATOR.VERSION:4.12.0-BETA
;GENERATOR.BUILD_DATE:2021-10-22
;TARGET_MACHINE.NAME:Ultimaker 2+ Connect
;EXTRUDER_TRAIN.0.INITIAL_TEMPERATURE:260
;EXTRUDER_TRAIN.0.MATERIAL.VOLUME_USED:21423
;EXTRUDER_TRAIN.0.MATERIAL.GUID:a17b348e-9e68-435a-bdeb-ee933c9193d5
;EXTRUDER_TRAIN.0.NOZZLE.DIAMETER:0.4
;EXTRUDER_TRAIN.0.NOZZLE.NAME:unknown
;BUILD_PLATE.TYPE:glass
;BUILD_PLATE.INITIAL_TEMPERATURE:100
;PRINT.TIME:15742
;PRINT.GROUPS:1
;PRINT.SIZE.MIN.X:47.536
;PRINT.SIZE.MIN.Y:56.366
;PRINT.SIZE.MIN.Z:0.2
;PRINT.SIZE.MAX.X:189.117
;PRINT.SIZE.MAX.Y:163.936
;PRINT.SIZE.MAX.Z:10.55
;END_OF_HEADER
;Generated with Cura_SteamEngine 4.12.0-BETA
T0
M82 ;absolute extrusion mode

G92 E0
M190 S100
M109 S260
G280 S1
G0 Z20.001
G1 F1500 E-6.5
;LAYER_COUNT:70
;LAYER:0
M107

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7566003/cura.log)

fvrmr, 2021-11-19:
Hi @rsa2010 thank you for your report. 
It is showing in the gcode here: ;BUILD_PLATE.INITIAL_TEMPERATURE:100
Was this also the case for previous versions?

rsa2010, 2021-11-20:
@fvrmr Not sure since I have bought the printer recently and alsmost immediately upgraded Cura.
"It is showing in the gcode here: ;BUILD_PLATE.INITIAL_TEMPERATURE:100" is in the header info but the G-code (printer instruction) is missing. (and the bed is not heating up unless I manualy add the G-code).
Printer does not wait for bed temp but I guess that's UM2C issue ?

rsa2010, 2021-11-22:
I'm pretty sure it is a defective printer. closing this issue

closed 2021-11-22 by rsa2010