Ultimaker/Cura #10878: Possible problem with baud rate detection

hadmut, 2021-11-18:
### Application Version

4.12.0

### Platform

Ubuntu 20.04 + snap

### Printer

Longer Cube 2

### Reproduction steps

Cura 4.4.1 (Ubuntu package that comes with Ubuntu 20.04) could easily print the sliced data directly to the printer over usb. 

Cura 4.12.0 (Comes as a snap instead of old fashioned ubuntu package) can't and hangs after choosing to print over USB. Cura believes to be printing, but can't even show the temperature. Nothing happens. 

### Actual results

Does not print on USB printer

Throws exception instead

### Expected results

Should print on USB printer as cura 4.4.1 did. 

Should not suffer from exception

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Observation 1:

Cura 4.12.0 tries to guess/detect the baud rate. I'm not sure, what's the correct one, probably 115200, but cura sometimes comes to the conclusion that it is 250000. I don't think that the printer would choose different baud rates. 

Since trying wrong baud rates can result in random nonsense data beeing sent, this detection test could possibly drive the printer into some  confusion. 

The printer profiles should contain baud rate settings, and if set, not guess the baud rate. 

Observation 2:

Error message on stdout/stderrr:

DEBUG:FileLogger:[MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [764]: Slicing took 1.3641674518585205 seconds
DEBUG:FileLogger:[MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [765]: Number of models per buildplate: {0: 1}
DEBUG:FileLogger:[MainThread] CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [779]: See if there is more to slice...
DEBUG:FileLogger:[MainThread] UM.Backend.Backend._logSocketState [196]: Socket state changed to Closing
DEBUG:FileLogger:[MainThread] UM.Backend.Backend._onSocketError [220]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
DEBUG:FileLogger:[MainThread] UM.Backend.Backend._logSocketState [198]: Socket state changed to Closed
DEBUG:FileLogger:[MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [959]: Backend finished slicing. Resetting process and socket.
DEBUG:FileLogger:[JobQueueWorker [4]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/ttyUSB0 if baud rate 115200 works. Retry nr: 0. Wait timeout: 3
DEBUG:FileLogger:[JobQueueWorker [4]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/ttyUSB0 if baud rate 250000 works. Retry nr: 0. Wait timeout: 3
DEBUG:FileLogger:[MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.02257704734802246 seconds
DEBUG:FileLogger:[JobQueueWorker [4]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/ttyUSB0 if baud rate 500000 works. Retry nr: 0. Wait timeout: 3
DEBUG:FileLogger:[JobQueueWorker [4]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/ttyUSB0 if baud rate 230400 works. Retry nr: 0. Wait timeout: 3
DEBUG:FileLogger:[JobQueueWorker [4]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/ttyUSB0 if baud rate 76800 works. Retry nr: 0. Wait timeout: 3
DEBUG:FileLogger:[JobQueueWorker [4]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/ttyUSB0 if baud rate 57600 works. Retry nr: 0. Wait timeout: 3
DEBUG:FileLogger:[JobQueueWorker [4]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/ttyUSB0 if baud rate 38400 works. Retry nr: 0. Wait timeout: 3
DEBUG:FileLogger:[JobQueueWorker [4]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/ttyUSB0 if baud rate 19200 works. Retry nr: 0. Wait timeout: 3
DEBUG:FileLogger:[JobQueueWorker [4]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/ttyUSB0 if baud rate 9600 works. Retry nr: 0. Wait timeout: 3
DEBUG:FileLogger:[MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.011644601821899414 seconds
DEBUG:FileLogger:[JobQueueWorker [4]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/ttyUSB0 if baud rate 115200 works. Retry nr: 1. Wait timeout: 15
DEBUG:FileLogger:[JobQueueWorker [4]] USBPrinting.AutoDetectBaudJob.run [79]: Detected baud rate 115200 on serial /dev/ttyUSB0 on retry 1 with after 3.01 seconds.
Exception in thread USBPrinterUpdate:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/snap/cura-slicer/54/usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py", line 280, in _update
    if re.search(r"[B|T\d*]: ?\d+\.?\d*", line):  # Temperature message. 'T:' for extruder and 'B:' for bed
  File "/usr/lib/python3.8/re.py", line 201, in search
    return _compile(pattern, flags).search(string)
TypeError: cannot use a string pattern on a bytes-like object
DEBUG:FileLogger:[MainThread] cura.AutoSave._onTimeout [64]: Autosaving preferences, instances and profiles took 0.00927591323852539 seconds

possibly related to Observation 1 and possible wrong Baud rate.

GregValiant, 2021-11-19:
Take a look at [THIS POST](https://community.ultimaker.com/topic/33214-cura-not-connecting-to-printer/page/2/) over on the Ultimaker Forum.  It is in regards to a Longer LK4 Pro and the USB to Serial driver on the computer but may apply to other Longer printers.  The page previous to that one has additional things to check.

fvrmr, 2021-11-19:
Hi @hadmut thank you for your report. 
Cura doesn't support and maintain USB printing since our printers doesn't use this anymore. And we can't test this for third party printers
Also your report is about a snap package, which is not the official AppImage of Cura. You can find the AppImage here: https://ultimaker.com/software/ultimaker-cura

Ghostkeeper, 2021-11-26:
The code there is indeed wrong, but it's not Ultimaker Cura's source code. We use `b"..."`, not `r"..."`:

https://github.com/Ultimaker/Cura/blob/9a4836a7b082c7778d1fccea51c9d221c18ccf7b/plugins/USBPrinting/USBPrinterOutputDevice.py#L280

It's been `b"..."` ever since that check got introduced. That would indeed give a `TypeError: cannot use a string pattern on a bytes-like object.`

I think whoever packaged that made a patch to change Cura's source code, but made a mistake here and didn't test their patch well.

Ghostkeeper, 2021-11-26:
The patch is made here, and should be fixed here: https://github.com/glasen/snap-cura/blob/0f08ae9156ed5e0dabdc2d885a0dad63517de598/patches/cura/cura.patch#L248

closed 2021-11-26 by Ghostkeeper