Ultimaker/Cura #10879: MKS WiFi Plugin

1lug1998, 2021-11-19:
### Application Version

4.12.0

### Platform

Windows 11

### Printer

Custom

### Reproduction steps

Cura no longer uploads proper gcode to MKS wifi plugin after update to 4.12. When viewing the files on the SD card after trying to upload, they have no file extension. 

### Actual results

The files have no file extension. 

### Expected results

The files should have .gcode after them. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I can't submit a file with no file extension

fieldOfView, 2021-11-19:
Issues with the MKS Wifi Plugin should be reported on their issue queue instead of here. See https://github.com/Jeredian/mks-wifi-plugin/issues

fvrmr, 2021-11-19:
Hi @1lug1998 thank you for your report. 
Like @fieldOfView mentioned please report your issue to the plugin developer. 

I will close this issue since we won't fix this.

closed 2021-11-19 by fvrmr