Ultimaker/Cura #10881: SolidWorks plugin

samirabaza, 2021-11-19:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Prusa I3 mk3

### Reproduction steps

parts made by SolidWorks 2021
are not opening in Cura 
even older files are not working
it takes forever to load or have error 

### Actual results

The Solidworks file does not open.
Solidworks installed on the device

### Expected results

the file should be loaded

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

![Screenshot 2021-11-19 192340](https://user-images.githubusercontent.com/9566057/142657014-e12cc610-7d1f-4fc2-83ea-136a91133c98.jpg
[image]
)

nallath, 2021-11-20:
Could you also provide the log files? There is a small chance that this is an issue of Cura itself, but it's more likely an issue caused by the plugin that you are using. If that is the case, you will need to contact the developer of that plugin. 

Ultimaker also has a native cad plugin, but that is part of our subscription model.

fvrmr, 2021-11-22:
See also: https://github.com/Ultimaker/Cura/issues/10806
Closing this issue since this is a duplicate. There you can also find temporary a work around.

closed 2021-11-22 by fvrmr