Ultimaker/Cura #10882: Visualize more slicing decisions:  retraction, retraction skipped because limits hit, wipe, coasting, etc.

hariedo, 2021-11-19:
### Is your feature request related to a problem?

It can be difficult to understand what the effects of various slicing options do, or how often various options apply to the gcode when turned on or off.

For instance, there are many different interrelated limits on when retraction can be used or can be blocked, but the only way to really get a sense of whether there's any effect is to reprint a small test repeatedly to see how it comes out. Filament crushing/grinding has several causes which are affected by part geometry and these slicing limit options.

Relating printing problems by looking at the ruined printed part and the settings is not always easy, especially when the slicer has to decide to use or skip certain gcode commands, but does not log any of these anywhere but the resulting gcode.

### Describe the solution you'd like

I think the best way to support this kind of visualization is to add another "color scheme".  For instance, duplicate the code used for the "Line Type" scheme, but hide most of the travels.  Showe the places where retractions occur, and opportunities for retraction which weren't executed because of the various max travel or min travel requirements, and wipes and coasting.

An alternative would be to add more toggles for kinds of line types (where "retraction" might be rendered similar to "starts" and "coasting" might be rendered similar to "travels".

It might also be nice to have a text area that gives some additional summary facts like Number of Retractions, Most Retractions Performed Within Window, and Number of Times Max Retractions Within Window Reached, but the visualization is more important to debug specific issues where filament is getting crushed, features are getting welded, or holes are appearing in walls.

### Describe alternatives you've considered

The kinds of slicing decisions I want to visualize are typically discarded runtime information, like "no, we need to skip this retraction because constraint XYZ says so."  Even if those runtime decisions were logged in comments in the gcode, it's hard to relate them to specific areas of the geometry which might be able to be redesigned to avoid problems.

Example: a print fails whenever the bunny head transitions into the bunny ears... oh, there seem to be 120 retractions within 4mm of filament, so the extruder is crushing the filament.  Or the print succeeds but heavy stringing between the bunny ears does not seem to be related to temperature... oh, the settings to control filament crushing kicked in too early, so a retraction-heavy area did not get the benefits of retraction.

### Affected users and/or printers

Visualization is important to debug specific issues where filament is getting crushed, features are getting welded, or holes are appearing in walls. The options to control these situations have an impact on the gcode, but it's pretty opaque to diagnose without trial-and-error reprinting.

fvrmr, 2021-11-23:
Hi @hariedo thank you for your feature request.
I have discussed this with my colleagues and we decided to defer this. 
The retraction is visible in preview mode with line type with travels. You can see dark blue and light blue for retractions.
The retraction skipped because limit hit is a bit complex to implement because this is something we don't show in the gcode currently so if we want this we have to change that. 
And the others don't have a priority for us to implement. I hope you understand.

daniels220, 2022-01-07:
Is there documentation about the line colors? I wondered about the dark vs. light blue travel lines but saw no explanation. Maybe adding an indented checkbox for "color retracted travels separately" (but that wouldn't fit, so maybe something a little different)?

Also, travel lines can easily vanish against the backdrop of the actual print lines, the subtle difference in color even more so—would be nice to have the option to show a bright obvious indicator for retractions, specifically—something similar to the "start" dots, in a different color or shape.

Ghostkeeper, 2022-01-14:
The explanation is supposed to be in the legend of the simulation view. It's not entirely complete though: Support interface and retracted travel moves are missing from the list (but are still disabled when disabling respectively the support or the travels).

daniels220, 2023-06-29:
I absolutely do think this is important and would improve how I use Cura. I've played with PrusaSlicer since my last comment, and they do a _better_ job of displaying retractions/starts/stops—at least bringing Cura up to that standard would be great, and additional info about why a particular travel move was/was not retracted and does/does not use combing would be helpful as well. Making individual lines clickable with a popup of specific info would be amazing actually but outside the scope of this particular feature I think.