Ultimaker/Cura #10884: [FR] Replacement patterns for print area

ChipCE, 2021-11-19:
### Is your feature request related to a problem?

Hello,
I'm making a macro that will dynamically change the bed_mesh area based on the size of the printed part. So the fw will only probe on the area that the part will be printed. To make that possible, the slicer need to parse the print area parameter via gcode to the fw.
[Sample video](https://www.youtube.com/watch?v=ag6nWA3_scQ)

### Describe the solution you'd like

As I looked in the Cura's exported gcode, there are already info about print area. It is possible to add replacement patterns (ex: {min_x}) for that data?

Current Cura's exported gcode header:
<pre>
;MINX:47.454
;MINY:54.449
..
;MAXX:115.788
;MAXY:127.504
...
</pre>

Thank you!

### Describe alternatives you've considered

none

### Affected users and/or printers

Any printer running firmware that support macro.

fvrmr, 2021-11-23:
Hi @ChipCE thank you for your feature request.
I noticed in the description of your linked video it shows that this is already working with a plugin is that correct?

ChipCE, 2021-11-23:
Yes, the plugin use regex to search string pattern and replace specified string in gcode make it only work in some specified case. Since the data already there, is is possible to make it easier to use, not just the area bed mesh but also other macro? As I known, most popular slicers already have that option.

dapostol73, 2021-11-25:
This would be great addition as I would really like to be able to use these variables.  Ideally, it would be nice that all of these are exposed as properties.  

TIME:  {print_time}
MINX: {print_min_x}
MINY: {print_min_y}
MINZ: {print_min_z}
MAXX: {print_max_x}
MAXY: {print_max_y}
MAXZ: {print_max_z}

Ghostkeeper, 2021-11-26:
Duplicate of https://github.com/Ultimaker/Cura/issues/7270.

closed 2021-11-26 by Ghostkeeper