Ultimaker/Cura #10885: Improved prime tower

Juusoz, 2021-11-19:
### Is your feature request related to a problem?

The prime tower in Cura is currently really wasteful. It purges on layers layers that do not have a tool change, for example. This gets expensive fast if you're using more expensive materials like PVA, for example.

In this example picture, more PVA is being used on the purge tower than PLA on the entire actual print, despite PVA only being used in a couple of spots.

[image]

### Describe the solution you'd like

There are a couple of things that can help with this:
1. **Skip purging on layers that do not have a tool switch**. PrusaSlicer and SuperSlicer have this. And Cura does have the "Gantry Height" machine setting that would make keeping track of whether or not it can skip the purge layer really easy.

[image]

2. **Print infill (and supports) with purge**. Using the purge for the infill of the model would open up the opportunity to not do full, solid layers in the purge tower. You could have a variable infill density in the purge tower to match the material it does not need to purge.

[image]

### Describe alternatives you've considered

I've completely minimized the amount of purging I need to do using base Cura, but it still is not enough. I'm considering switching to SuperSlicer because their purge tower is better than yours.

### Affected users and/or printers

All printers with 2 to 1 material nozzles will benefit from this.

fvrmr, 2021-11-22:
Hi @Juusoz thank you for your request.
You can disable Wipe Inactive Nozzle on Prime Tower. But I noticed that you are using PVA on extruder one and extruder one will always extrude. If you switch your extruders that should help a bit. 

About the second point, see https://github.com/Ultimaker/Cura/issues/3657

Your third point will be bring up for discussion with the profile and materials team. Keep you posted.

Juusoz, 2021-11-23:
>But I noticed that you are using PVA on extruder one and extruder one will always extrude. If you switch your extruders that should help a bit.

Yeah, I noticed that, but unfortunately it won't be possible in my situation. You see, I have two extruders on my printer, one on the left side of the printer and one on top of it. And my PVA is stored in a fairly large drybox to the left of it, so I literally am not able to put the PVA into the top extruder without modifying the printer.

[image]

I don't know how hard it would be to implement, but it would be nice to be able to pick which extruder to make the prime tower shell with.

Although, then again, this might just be a me problem.

Ghostkeeper, 2021-11-26:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/5282. Please continue the conversation there.

Any other issues should be made into separate tickets. It's impossible to track the status of them otherwise.

Ghostkeeper, 2021-11-26:
> extruder one will always extrude. If you switch your extruders that should help a bit.

This is not entirely true. It will select the material with the greatest Adhesion Tendency as the one to use for the outer shell of the prime tower. For PLA this setting is greater than for PVA, so I'd expect the PLA extruder to still be used for the outer shell.

closed 2021-11-26 by Ghostkeeper