Ultimaker/Cura #10886: Slice errors and imported objects not centered after update to 4.12.1

vicasta01, 2021-11-19:
### Application Version

4.12.1

### Platform

macOS Monterey

### Printer

Longer LK4

### Reproduction steps

1. Update Cura 4.12.0 to 4.12.1 by replacing old Application as usual. 
2. Import an STL file to Cura.
3. Object imported would be centered on print bed after import.
4. Click on "Slice" 
5. Multiple errors appear. 

### Actual results

Imported object is placed to the right and off the bed. 
Multiple errors prevent slicing.
[image]
[image]
 

### Expected results

(((SCREENSHOTS FROM CURA 4.12.0)))
Imported object is placed at the center of the print bed.
Model is sliced correctly without errors
[image]
[image]
.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Problem semi-solved by reverting from Cura 4.12.1 to 4.12.0 by replacing the App on the Applications folder.

fvrmr, 2021-11-22:
Hi @vicasta01 thank you for your report.
Looking at your models you are printing your models one at a time. It that correct?
I tried both on 4.12 and 4.12.1 on Monterey and I could slice multiple objects with one at a time mode.
I think your issue is that the models are overlapping to much for printing one at the time. 

You could try to save your project in 4.12 (the one that slices correctly) as a project file. And then open that project file in 4.12.1 to see if it slices then. 
Let me know if that works!

vicasta01, 2021-12-03:
Hi @fvrmr !
Yes, I'm printing one at a time, they're not too tall, so height clearance is not a problem, Cura 4.12.0 sliced this particular setup just fine and I actually printed it no problem. Exported as a project as you suggested, then imported the project into 4.12.1, "Unable to slice". Double checked printer settings in 4.12.1 and they're the same as in 4.12.0, which are the defaults for my printer, nothing tweaked or changed.
[image]

Ghostkeeper, 2021-12-15:
When printing in one-at-a-time mode, the printer will need some horizontal clearance too, to prevent hitting the previously printed object when printing the next one. For this, the object has two collision areas: One where the print is, and one where the print head can be while printing it. The print head (larger one) may not hit the print (smaller one) of a different object.

Cura will optimise the order in which these objects are printed to allow the print head collision area to be as small as possible. The order is optimised when the slice starts.

In your screenshots it can be seen that the print head would indeed hit the other objects. As a result, Cura is unable to create an ordering in which they can safely be printed. This prevents it from starting the slice.

It's expected that it won't slice this way. You need more clearance.

But what I do think should happen here is to clarify to the user why it can't slice.

vicasta01, 2021-12-15:
> When printing in one-at-a-time mode, the printer will need some horizontal clearance too, to prevent hitting the previously printed object when printing the next one. For this, the object has two [...]

Was anything changed in this regard in the 4.12.1 update? Asking this because I’ve printed similar things with 3 different printers and everything sliced and printed okay all the way back from cura 4.10 to 4.12.0, and the problems started only with the latest Cura. Thanks for your input and help!

tablatronix, 2021-12-16:
Is this the same? Its very annoying, models are not auto dropped on bed or centered. Most of my models wont even go above the bed, I have to manually raise them and drop them or else they are -z. They also seem to be always badly aligned on faces, eg upside down! Printing anything seems to take forever to just get them positioned.

[image]

Ghostkeeper, 2021-12-27:
> Is this the same?

I think you have a different issue. Perhaps you're trying to import 3MF files and wondering why the models are placed in the location specified by the 3MF file?

When models are (partially) outside of the build plate, the drop to build plate is no longer applied.

GregValiant, 2024-11-16:
Is this still a problem in current versions of Cura (5.8.0 and up)?  Can this be closed?

GregValiant, 2024-11-16:
Done.

closed 2024-11-16 by GregValiant