Ultimaker/Cura #10889: Stop bleeding @ measurement

DrStrixx, 2021-11-20:
### Is your feature request related to a problem?

The nozzel bleeds a lot of filament when it measures the hight of the table.
And then I have to clean the nozzel from filament when it starts to print! Or it will affect the print when it get stuck in parts.

### Describe the solution you'd like

Let the nozzel stay @ 150 degrees untill it have measure all points, and then pause over "home" until it reaches the right temperatur.
I did use Prusa slicer and it does like that. No bleeding @ all

### Describe alternatives you've considered

Any way to set temp to 150 before print

### Affected users and/or printers

Bleeding from nozzel

GregValiant, 2021-11-20:
That is all a function of your StartUp Gcode.  Something like this will work.
M104 S150 ;start heating the hot end 
M190 S{material_bed_temperature_layer_0} ;heat the bed and wait for it to get to initial layer temperature
M109 S150 ;wait for the hot end to get to 150° just in case.
G28 ;Auto Home
G29 ;Auto Level
G1 F2700 Z5 ;raise the Z
G1 F3600 X0 Y0 ; Go to 0,0,5
M109 S{material_print_temperature_layer_0} ;heat the hot end to initial layer print temperature and wait for it.

DrStrixx, 2021-11-21:
Ok, great, but how do I implement it in the gcode as a standard settings to every print?
I am not finding any way to do it in Cura.

GregValiant, 2021-11-21:
In the Prepare screen click on your printer name.
Click on Manage Printers.
Click on Machine Settings.
In the "Start-Up G-Code" box are the commands that run every print before the print actually starts.  That's where you make the change.  Unfortunately, you didn't include your printer model in the request.  That would be good to know as some printers handle this differently.

The following would work on my Ender 3 Pro (G-code lines that start with a semi-colon are comments).  You will notice that the Auto-Level command I show is G29.   Yours might be M420.  You would need to make adjustment.  Also, the purge lines go to Y200.  If you have a smaller machine it would still try to get there and make horrible complaining noises when it stops because it can't go any further.
;
; Ender 3 Custom Start G-code
;
G90 ;absolute movement
G21 ;metric units
M220 S100 ;set the feedrate to 100%
M221 S100 ;set the flowrate to 100%
M140 S{material_bed_temperature_layer_0} ;start heating the bed
M109 S150 ;heat the hot end to 150 and wait for it
M190 S{material_bed_temperature_layer_0} ;wait for the bed to hit your initial layer bed temperature
G92 E0 ; Reset the extruder
G28 ; Auto-Home all axes
G29 ;Auto-Level
M104 S{material_print_temperature_layer_0} ;start heating to Initial Layer Print Temperature
G1 F2700 Z10 ; make sure the nozzle is above the build surface
G1 F3600 X0 Y0 ; move to the left front corner
M109 S{material_print_temperature_layer_0} ; heat and wait to hit the temperature.
M82 ;Absolute extrusion mode
;    Purge lines
G1 X1.0 Y20 Z0.3 F5000.0 ; Move to start
G1 X1.0 Y200.0 Z0.3 F2100 E15 ; Draw the first line 35mm/sec
G1 X3.0 Y200.0 Z0.3 F5000 ; Move
G1 X3.0 Y20 Z0.3 F2100 E30 ; Draw the second line 35mm/sec
G1 E28 F1800 ;retract 2mm
;   End of purge
G92 E0 ; Reset Extruder
G1 Z10.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X10 Y20 F5000.0 ; Move over to prevent blob squish
;
;End of Start-Up G-code
;

DrStrixx, 2021-11-21:
Oh... Thanks!

This is my "Start Code"

M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration

M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate

M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration

M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk

M220 S100 ;Reset Feedrate

M221 S100 ;Reset Flowrate

And this is the "End code"

G91 ;Relative positioning

G1 E-2 F2700 ;Retract a bit

G1 E-2 Z0.2 F2400 ;Retract and raise Z

G1 X5 Y5 F3000 ;Wipe out

G1 Z10 ;Raise Z more

G90 ;Absolute positioning

Can u help me with this

I am lost in coding scripts... Have no clue what to do...

Regards

DrStrixx, 2021-11-21:
Forgot!

I have A Creality cr10s-pro

GregValiant, 2021-11-21:
When I add a CR10sPro in Cura, the startup gcode is much different than what you show.  The snippet below has been altered using the CR10sPro Start-Up G-code as a template.  You can copy and paste it into your Start-Up G-code box in Cura and over-write what is there.  Below that is an End G-code that should work as well.    I would suggest slicing a calibration cube and printing it after you make these changes.  You may want to make an additional tweak or two.
;
; Custom CR-10S Pro Start-Up G-code
;
G90 ;absolute movement
G21 ;metric units
M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

M140 S{material_bed_temperature_layer_0} ;start heating the bed
M109 S150 ;heat the hot end to 150 and wait for it
M190 S{material_bed_temperature_layer_0} ;wait for the bed to hit your initial layer bed temperature
G92 E0 ; Reset the extruder
G28 ; Auto-Home all axes
G29 ;Auto-Level
M104 S{material_print_temperature_layer_0} ;start heating to Initial Layer Print Temperature
G1 F2700 Z10 ; make sure the nozzle is above the build surface
G1 F3600 X0 Y0 ; move to the left front corner
M109 S{material_print_temperature_layer_0} ; keep heating and wait to hit the temperature.
M82 ;Absolute extrusion mode
; Purge lines
G1 X1.0 Y20 Z0.3 F5000.0 ; Move to start
G1 X1.0 Y200.0 Z0.3 F2100 E15 ; Draw the first line 35mm/sec
G1 X3.0 Y200.0 Z0.3 F5000 ; Move
G1 X3.0 Y20 Z0.3 F2100 E30 ; Draw the second line 35mm/sec
G1 E28 F1800 ;retract 2mm
; End of purge
G92 E0 ; Reset Extruder
G1 Z10.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X10 Y20 F5000.0 ; Move over to prevent blob squish
;
;End of Start-Up G-code
;

;
;  CR10 S Pro Custom End Gcode
;
G91 ;Relative positioning
G1 E-2 F2700 ;Retract a bit
G1 E-2 Z0.2 F2400 ;Retract and raise Z
G1 X5 Y5 F3000 ;Wipe out
G1 Z10 ;Raise Z more
G90 ;Absolute positioning

G1 X0 Y{machine_depth} ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed
M220 S100
M221 S100
M84 X Y E ;Disable all steppers but Z
;
; CR10 S Pro

DrStrixx, 2021-11-21:
Awesome, thanks!

DrStrixx, 2021-11-21:
Noticed that I could scroll down in codes windows! And I think I missed
that before..... And there were more codes then I copied to u. Silly me...
🤓

GregValiant, 2021-11-21:
Yeah, I figured there should have been more code in both the boxes.

fvrmr, 2021-11-22:
Hi @DrStrixx glad to see that you have made it work with help of @GregValiant 
I will close this issue since this is probably solving your issue. But please let us know if you have more questions!

closed 2021-11-22 by fvrmr