Ultimaker/Cura #10891: Fail to slice a 200mm object with build plate 220mm

billygr, 2021-11-20:
### Application Version

4.12

### Platform

MacOS

### Printer

Ender 3 V2

### Reproduction steps

Cura 4.12 fails to slice an object of 200mm while using a build plate (default of 220x220) on Ender 3 (actually Ender 3 V2) .

It present it as it doesn't fit on the build volume.

Relevant project file is attached

### Actual results

Object fits and can be sliced with PrucaSlicer 2.3.0 . The case here is to sort it out with Cura and not using a different slicer

### Expected results

Be able to slice it since it fits the build volume

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Cura_4.12_object_200mm.zip](https://github.com/Ultimaker/Cura/files/7575446/Cura_4.12_object_200mm.zip)

GregValiant, 2021-11-20:
Cura takes into consideration the area that the brim/skirt/raft will take up.  Change your "Skirt Distance" to 2mm and the gray areas will get smaller allowing the part to slice.  You could also rotate that part 45°.
200mm part + 10mm skirt distance on top + 10mm skirt distance on bottom + 2.4mm of skirt = 222.4 which is more than your Max Y.
You can Auto-Home and then use the knob to move the nozzle 2.5mm from the left edge and 2.5mm from the front edge and the Z at 0.  Then hit Set Home Offsets in the LCD menu and then Save Settings.  In Cura Machine Settings you can then change the build plate size to 230 x 230.  That will leave a 2.5mm safety zone all around the build plate and will maximize your printable area.

billygr, 2021-11-21:
Hi Greg,

I have tried to change the build plate size to 230x230 before submitting the bug report but still even with the 222.4 as total print needed it fails to slice.

I need to reduce to skirt size to 6mm to make it fit. 200+6+6 + 2.4 = 214.4 mm. (keeping the build plate at 230x230) so somewhere it is being calculated differently

The reason that i try to keep it to close to the defaults (without manual changes) is that i have hitted in the past the clip on the build plate resulting in a bit bended axis

GregValiant, 2021-11-21:
Do you have a gray area in the front and back of the Cura build plate that never changes size?  Creality added the "bed clip" disallowed areas to the Ender 3 definition in Cura.  Those areas are not adjustable.  They can be removed by altering the definition file, or you can install an Ender 3 Pro in Cura and use it.  The 3 Pro definition doesn't have the clip areas but is otherwise the same as the regular Ender 3 definition.
The easiest thing to do is to rotate the part so it lays kitty-corner.

billygr, 2021-11-21:
Thanks !!!. That explains it. Front and Back of the cura build plate have a gray area. 
I will do the changes in the definition file, no need to keep the bug open, at least it is now documented (and understood)

GregValiant, 2021-11-21:
When you update Cura to a new version you will lose your "customization".  I keep a separate folder for files I've customized and after updating Cura I copy and paste them into the new Cura folders.

closed 2021-11-21 by nallath

Ghostkeeper, 2021-11-26:
The Ender 3 Pro is a variant of that definition without the clip areas (normally because that printer has a magnetic bed).