Ultimaker/Cura #10892: G code generataing bug

Zasti78, 2021-11-21:
### Application Version

4.12.1

### Platform

Win 10

### Printer

Artillery x2

### Reproduction steps

Hi Guys. My Cura has been updated yesterday morning, The first(a bigger print) generated by new versioned cura and my old profil)l, It was a Bowl.(ower 7h printing time) I left the job came back later and was seeing, the printer behaves strangly.was shaking like an idiot. The noze touched the printed surface,(look like all machine is owerdriven). I took the speed to 40% and the extruder speed to 70%, . the job done.(but isnt beutiful)
Morning my first step is check the code, what is strange. the G0 and the G1 alternate line by line  throughout the  code with differerts speed rate.
G0 and F?
In my profil the printer is artillery x1 witch is recomended by wizzard,(and its working over months) In the machine settings at the "start g code " and "end G-code" there is no such definition. 

"G0 F7500 X92.93 Y170.283
G1 F1500 X140.263 Y122.95 E848.14567
G0 F7500 X140.604 Y123.231
G1 F1500 X93.2 Y170.634 E850.59835"
 Center of the G code.
[image]

I  tried to rgenerate a G-code ( today morning) again
But after opened stl model I cannt do anymore.
Doesnt matter cusom or recomended settings, the cura doesnt slice anymore!I have no option to slice only to save(the nothing) to the disk or removable media.
It gives me  "0 mins 0g.0.00m"
Tired many models many times , same result.
[image]

### Actual results

I tired to reset my profil,  set a new machine by wizzard. (install reinstall etc)
(same bowl, same cura, recomended settings)
[image]

I can slice now, but the g-code is looks same.
There are a lots of G0 commands with feed rate in code.
[image]
Im going to swap back to the 4.11 cura. 

### Expected results

xy

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Cura log

[cura.zip](https://github.com/Ultimaker/Cura/files/7576075/cura.zip)
and (I know but...) the before and after g -code-
[G-codes.zip](https://github.com/Ultimaker/Cura/files/7576071/G-codes.zip)

Asterchades, 2021-11-21:
G0 and G1 are functionally identical on any machine that is not SCARA-based. It's not uncommon, however, for travel movements (ie Movements without extrusion) to be done using G0 and printing movements (ie Movements _with_ extrusion) to be G1. This is even recommended by the Marlin devs for cross-compatibility with laser and routing machines. Looking through the captures you've provided shows they're both being used correctly, with E values only on G1 lines.

Looking through your snippets it appears you simply have different speeds set for travel and printing, which is to be expected. The first in any sequence of G0 commands (ie Travel moves) sets the feed rate to F9000 (150mm/s), while your actual printing commands set it to either F3600 (60mm/s) or F1800 (30mm/s) presumably based on what feature is being printed. While Marlin 2 allows for G0 and G1 to have different feed rates set, Cura is still outputting Marlin 1 code for compatibility - this hurts nothing other than the GCode file size.

I'm unsure as to why your project did not slice, nor am I sure why the upper snippet occurred. Your upper snippet appears to be fine gap fill but is maintaining a constant direction rather than alternating, so something is interfering with it (perhaps related to Monotonic settings?).

Zasti78, 2021-11-21:
Hi
I got it.( around G-code.)
Did you check the log file?
abouth machine settings.
What did i messed up?
I think (99%) i had here the same settings, same as in cura 4.11.x
This print was the first job with the updated cura.

//If i want to test or swap example new material, nose etc, i am setting up a new profil, with new machine, or new materials.//
and i had in this bowl print  over extrusion problem also,
 im sure i did never change in setting the exrusion's speed example, the material was a "generic PLA" 
I uninstalled it, and installed the 4.11.0
I am going to try it again at the next weekend when i have no urgent prints, 
Let's see if the problem comes up again
Thanks for explanation.
R: ZAS

fvrmr, 2021-11-22:
Hi @Zasti78 thank you for your report. 
Could you also share your project file (file - save project)?

You could also try to open your project file from 4.11 in 4.12. Because that should give the same results.

Zasti78, 2021-11-22:
Hi, 
Unfortunately (now) i have deleted all my profil, and started a new one, and reopened same project to play with settings. I have printed bowl yesterday and last night without any issue,
 so i do not have those settings anymore, and today i started to print ABS, so i messed up . but...
 I was exactly thinking about same, the project file, because it was open as a project file in the 4.12.
I am using more slicers, like prusa cura, win 3d builder, and just droping between those apps often. It can be the reason, but i have not any problem since t last weekend.(i've seen a cople of forums has topic about 4.12 had issues with profil from older versions.
So if iam ready with those prints , about friday or the weekend im going to try to reproduce, it i ll try to set how was, and give an update, than print. I have both exe files, so i can rewrite with exactly same 4.12 update.

Ghostkeeper, 2021-11-29:
I think Zasti is talking about the gap filling technique, shown here in layer view.
[image]

This indeed produces tiny line segments, separated by travel moves. So `G1 F1800` for the line segment (printed using the skin profile, 30mm/s) and `G0 F9000` (a travel move, at 150mm/s).

If settings are different, this may disappear, in particular if the wall line widths got changed or if Fill Gaps Between Walls is changed.

So this seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/5110 . The Arachne project intends to solve this issue permanently.

closed 2021-11-29 by Ghostkeeper

Zasti78, 2021-11-30:
HI 
Thank you

I installed the 4.12. and I just tried over and over again,( same profil from prev. version, same modell, about same settings. no problem at all,/Not even the empty g-code/
I could have waited before I swaped to 4.11,but I hurried.(i Still have a lots of printing work it was neccessary)
More than 2 days on 4.12,  11pcs models done 
and the machine does what it needs to.//I am beginning to believe that I may have made a mistake//

 I can guess what caused in that case that the machine behaved like idiot .

Thank you for taking the time to address this issue.

T: Zas