Ultimaker/Cura #10893: Documentation (popup help) missing critical info

gitcnd, 2021-11-21:
### Is your feature request related to a problem?

Don't know what "zero" means for "Z"  (e.g. the build plate itself? the bottom of the nozzle? ... you know that when "levelling", the nozzle never *touches* the plate at the end, right?)

### Describe the solution you'd like

The pop-up help for the item "Initial Layer Height" needs to properly explain what exactly that means, and how it is obtained (is this a setting that controls something, or is this a question asking me what my nozzle-bed gap actually is ??).

e.g. I've got a 0.3mm gap already - so - do I set this value to "0" (since I already have 0.3mm) or do I set it to 0.3mm (and CURA knows to use a Z of 0.0).

Or perhaps another option is needed?  "Initial Nozzle distance from Bed".

### Describe alternatives you've considered

Waste a bunch of time trying different numbers to see if I can work out what this number actually MEANS.

### Affected users and/or printers

Everyone

fvrmr, 2021-11-22:
Hi @gitcnd thank you for your request.
If you download the settings guide from the market place you will get an more extensive explanation
[image]

Here you can download the settings guide: https://marketplace.ultimaker.com/app/cura/plugins/Ghostkeeper/SettingsGuide2
So it has nothing to do with bed levelling. Let me know if this makes senses or if you have more questions.

gitcnd, 2021-11-22:
Notice how the guide itself is also missing the info I am talking about.

The only thing in the world that knows the "initial layer thickness" is the brain of the person who levelled the build plate - since they're the ones who know what gap was used.

e.g. I've got 0.2mm gap below my nozzle - so - when I select 0.2mm for "Initial Layer Height" - am I going to get 0.4mm (Cura's 0.2 plus my gap) or am I going to get 0.2mm ?

Like I said - this needs to be spelled out properly, and probably makes more sense to fix this "bug" by adding a new setting "nozzle-plate gap" so we can TELL Cura how much gap is already there...  (e.g. I set my gap to 0.3mm last time I was using a 0.6mm dia nozzle...).

Bottom line - something needs to be done so that layer zero is at Z=0.00 for me, and probably almost everyone else.

5axes, 2021-11-23:
In my case the initial bed level is around 0.03-0.05mm but this just because I m keep an initial flow level at 100%

gitcnd, 2021-11-23:
You mean 0.3mm, not 0.03mm, right?  0.03 is almost 4 times thinner than one sheet of paper - no *way* your bed is flat enough for that to work? - the nozzle is going to collide with it for *sure* - heck - heating probably moves it by more than 0.03mm.

5axes, 2021-11-23:
No it's correct **0.03 mm** and no there are no collision because this parameter "Initial Layer Height" is your first Z in the GCode. So in reality the distance between the nozzle tip and the Bed is equal (in my case) to : (First Layer Height - 0.03) => 0.2 - 0.03 = 0.17 mm.  (it's of course a theoretical point of view, as my printer and the deformation of my printing plate are not precise enough to respect this value on the whole printing area )

gitcnd, 2021-11-23:
OK - looks like this is definitely a bug then.  It does not let me put 0 for the initial layer height!

How is Cura supposed to know what rate to flow material out, when there's nothing in the settings to let it know the nozzle levelling clearance ?   Hoping that us users can "guess" this by kludging some extra-flow-rate for the first layer is crazy - and does not work anymore if we change nozzle sizes...

Getting this right is plain math - we should not be forced to guess numbers to get this to work - it should be programmed to work properly "out of the box" with how our printers are set up.  I dunno what kind of printer you use, but mine and all the others I know tell you to level the bed with a page of paper (about 0.1mm) between it and the nozzle.

5axes, 2021-11-23:
This is not a bug the Initial Layer Height is just **the height of the first layer** nothing to do with your bed level calibration. As you write in one of your previous comment No one knows how you level your Bed and witch method you rae using to do that. So it's normal that the first Layer height cannot be set to 0, this would mean that nothing would be extruded as the required height is zero.
[image]

gitcnd, 2021-11-23:
You're wrong.  The "requested height" can NEVER BE ZERO, because there is no situation where the nozzle is in contact with the build l plate (which would be zero).

Stop trying to win an argument, and think about this mistake more carefully: Cura needs to know how fast to extrude material for the first layer, and it can NEVER BE ABLE TO DO THAT unless there's someplace we told it what clearance is set when Z=0.0

Ghostkeeper, 2021-11-29:
Cura indeed assumes that when the g-code says to move to `Z0`, it would move the nozzle to touch the build plate. It doesn't normally move to `Z0` unless the start g-code specifies it (since that could damage the build plate). But it does base the amount of extruded material on `Z0` being the level of the build plate. This includes the assumption that the build plate is flat.

Of course, the build plate is not flat. This is one of the reasons why people make the initial layer height larger than others; it gives more clearance if the build plate goes a bit higher, and more tolerance for adhesion failure if the build plate goes lower.

Under that assumption, an initial layer height of 0 would give 0 extrusion, and can just be skipped then, causing the next layer to be the first layer. This would be identical to having the first layer height be equal to the normal layer height.

If you know that the surface of the build plate is consistently not at `Z0` and would like to compensate for it in software rather than in hardware, there are two things you could do:
* You could adjust the flow of the first layer directly to compensate, or
* you could use the [Z Offset Plugin](https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/ZOffsetPlugin) to directly move the entire print up or down.

Both can be used to the same effect, but the Z Offset Plugin is a bit easier to use if you know the offset you need to use.

The setting descriptions in Cura intend to be brief and mostly just descriptive rather than give printing tips. Printing tips are best left to other places, like Ultimaker's website, the Settings Guide or other online printing guides.

closed 2021-12-16 by no-response[bot]