Ultimaker/Cura #10897: ive been using cura a lot since i first got my 3d printer but im trying to print a wing for an rc airplane and i used the files that were provided with it in the prep menu didnt see anything wrong its not supposed  to have infil because there is a support pattern that comes with the wing to make eoom for wires and such but when i slice it it makes it completely hollow on the inside with no hole for wires or anything im not sure what the issue is

kayleb27, 2021-11-21:
### Application Version

4.12.0

### Platform

Windows 10

### Printer

creality ender 3

### Reproduction steps

I used Cura to slice a wing I was making for an airplane it already had support in it so i only put support on to make sure any over hangs would be okay bit it made it completely hollow without anything on the inside even though in the prep section it showed everything like it should but the software seemed to just say it was only going to print certain parts that it chose and I'm not sure why it wont even attempt to print certain parts or details 

### Actual results

i use 3d builder to view so i tried skipping that step incase it modified it did not work so then i put 0 infill tried different orientations didnt work but the files are good it comes with just gcode as well but it tells my printer to go very fast when it needs to be slow precise on the inside supports so im not sure why it is covering up all of the holes i tried with support and without same thing on both

### Expected results

so as you may be able to see in the pictures i provided that when its preparing it shows the hole through the middle for wires as well as some of the internal support but when sliced it puts a top and bottom on and makes it hollow on the inside and doesn't put a hole trough it[image]
[image]
[image]
[image]
[image]
[image]

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

if anymore info is needed please email me at kaylebbeam27@gmail.com

TheSkittle, 2021-11-21:
Can you please upload the project file? I can take a look at it and see why its not working. To be clear, that giant hole going up and down being covered up is the issue right?

fvrmr, 2021-11-22:
Hi @kayleb27 thank you for your report.
Like @TheSkittle mentioned a project file (file - save project) would be very helpful.
Also it could be the case that the missing pieces are to thin to print. You can check our Arachne release: https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_beta here we use variable line widths.

TheSkittle, 2021-11-22:
@fvrmr I've been hearing a lot about that thing. What is it? And what new features it have?

fvrmr, 2021-11-23:
@TheSkittle Arachne is a new engine that to fill available space better with walls, preventing gaps and improve paths by implementing variable line widths. This could help with printing thin walls like this model. 
You can download the beta here: https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_beta and we are currently working on improving this beta and will release a new build soon!

TheSkittle, 2021-11-23:
@fvrmr sounds cool. Also, when I print certain models, when the overhang becomes greater and greater gradually, it prints fine, but when the walls go inward gradually, the walls look like theres underextrusian , which is wierd because it only happens when the walls go inward,  not outwards, and I already have the flow rate on 150 percent

closed 2021-12-06 by no-response[bot]