Ultimaker/Cura #10898: Elephants foot offset

TheSkittle, 2021-11-21:
### Is your feature request related to a problem?

I make a model with multiple rings on the inside (such as a gyroscopic ring), and theres elephants foot. Since the rings are precise, when i use brim, the rings fuse together. When i use a raft, the bottom gets a terrible finish, and is all stringy and hard to separate the rings.

### Describe the solution you'd like

An elephants foot, offset.   An option to scale the bottom few (about 3) layers inward. This would make up for the elephants foot effect and would make it look as if theres no elephants foot at all. There should be 3 settings: 1. enabled? 2. layers 3. angle/offset

### Describe alternatives you've considered

rafts/brim. When making complex models with little clearance, brim would fuse them together and make it hard to remove the brim without slightly damaging the model. The raft would give the bottom a terrible rough,and stringy appearance.

### Affected users and/or printers

Anybody trying to make a complex model with little clearance, who elephants foot will greatly effect.

### Additional information & file uploads

I cant upload a picture because this setting doesnt exist.

fieldOfView, 2021-11-22:
Have you tried the "Initial Layer Horizontal Expansion" setting?

fvrmr, 2021-11-22:
Hi @TheSkittle thank you for your feature request.
This is a duplicate of: https://github.com/Ultimaker/Cura/issues/9613
We have ticket on our backlog for this.

I will close this issue since it is a duplicate!

closed 2021-11-22 by fvrmr

TheSkittle, 2021-11-22:
@fvrmr  sorry, i didnt realize. So in the end, was there a fix to this issue or a preexisting  feature for this?

fvrmr, 2021-11-23:
@TheSkittle no problem! It is still on our backlog as a new feature but it is not prioritized yet.

TheSkittle, 2021-11-23:
@fvrmr ok. I've seen mentions of a new software or something for cura. I forgot what the name was. Eternal or unreal or ethereum or something like that. What is it and what new features will It have