Ultimaker/Cura #10902: [Feature Request] Different icons for different filetypes

ilko-k, 2021-11-22:
### Is your feature request related to a problem?

No matter what filetype, once it is associated with Cura, they all have the same icon, making it difficult to distinguish them (i.e., opening a .gcode intead of .stl)

### Describe the solution you'd like

Creating different icons for each filetype that Cura actually supports.

### Describe alternatives you've considered

Nothing (yet)

### Affected users and/or printers

Everyone will benefit from this. This could eventually avoid users reporting false errors due to a careless mistake in dropping wrong files into Cura (i.e. when a gcode file is loaded, print setup is disabled and slicing obviously cannot be done but this can be misinterpreted).

Ghostkeeper, 2021-11-30:
They do get different icons for different file types on my computer, using Gnome. For STL I have a thumbnailer to make a 3D preview. For .gcode it categorises it as a text file. For 3MF it categorises it as an archive (because the 3MF file format uses the zip container). OBJ is blank.

Am I correct to assume that you meant on Windows? But on Windows, the icons it gives to the files in Explorer are the icon of the application assigned to be the default application for that file type. Cura doesn't really have control over them. Perhaps this should be customer feedback for Microsoft?

I believe (fact-check me on this) that MacOS will associate files with applications and display the application's icon similar to Windows. The difference being that on Windows this file association is by file extension rather than by MIME type. But also there we would have little control over what icons are being displayed, short of shipping 12 applications each with different icons.

So in short, I don't think there is really anything we can do within Cura's scope to change this.

ilko-k, 2021-11-30:
Hi, yes this is meant for Windows. Associating app's main icon to files is the default behaviour. But, archiver apps like 7zip have an icon library to differentiate file types. It's the idea.

fieldOfView, 2021-11-30:
> But on Windows, the icons it gives to the files in Explorer are the icon of the application assigned to be the default application for that file type. Cura doesn't really have control over them. 

Cura does not have control over them, but the installer does. If Cura.exe were to include multiple icons, the installer could set which icon to use for each of the filetypes it creates an association for. In https://github.com/Ultimaker/cura-build/blob/master/NSIS/FileAssociation.nsh#L140 you would need to change the `,0` index to point to the appropriate icon resource inside Cura.exe.

Ghostkeeper, 2021-12-13:
I see! That should indeed be possible then.

fvrmr, 2021-12-14:
We have discussed this feature request with the team and decided to create a ticket for this on our backlog. 
Devs see CURA-8761

ilko-k, 2021-12-14:
Interesting.

fvrmr, 2022-03-21:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features from our community and from Ultimaker, we have decided to remove this feature request from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this feature and create a pull request yourself.

closed 2023-07-13 by github-actions[bot]