Ultimaker/Cura #10904: Calculate Adaptive Layers independently for each object

Firedrops, 2021-11-23:
### Is your feature request related to a problem?

When printing multiple objects simultaneously, adaptive layers for each object is applied to all objects, making it essentially print at the finest resolution all the time.

### Describe the solution you'd like

Each object should have adaptive layers independently calculated and applied.

May also require enforced z hop and/or avoid objects during travel options.

### Describe alternatives you've considered

Splitting objects into separate prints, but this ends up taking a lot of extra time anyway due to heating/ABl cycles.

### Affected users and/or printers

Pretty much anyone who uses Adaptive Layers.

fvrmr, 2021-11-30:
Hi @Firedrops thank you for your feature request.
I will bring this up with the team. Keep you posted!

fvrmr, 2021-12-03:
I have discussed this request. And this is really complex to implement, because having multiple models with multiple layer heights could cause that the nozzle will bump into another object. That's why I will defer this issue. I hope you understand.

Firedrops, 2021-12-04:
Of course, as long as the idea's out there, hopefully someone can come up with a good workaround :)

mr-september, 2023-07-09:
I still think this is worth investigation. Due to issues raised above, implementing it may also need to touch "avoid printed parts when travelling" and/or z-hop algorithms/settings.

Firedrops, 2023-07-09:
I still think this is worth investigation. Due to issues raised above, implementing it may also need to touch "avoid printed parts when travelling" and/or z-hop algorithms/settings.