Ultimaker/Cura #10905: Cura 4.9-4.12.1 isn't responding after opening window.

digiTALker, 2021-11-23:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

RepRap

### Reproduction steps

1.Run Cura
2. Cura opened and not Responds.

### Actual results

When this happens does your computer works fine. Not responding just Cura.
Doesn`t matter logged in Ultimaker account  or not.
The problem starts after version 4.9. Since that time I used 4.8.
Interesting that first launch after installing works fine, untill login to Ultimaker account. Further even if logged out Cura is totally freezing.
I tried Cura 4.12 Beta and few times it works as expected. Than problem back again. If I run CuraCLI.exe some times it works good.

Also 4.9 - 4.12 beta works fine if close Autodesk Desktop Application and Autodesk licensing Servrer and Google Drive Desktop.
Unfortunately its not affect with 4.12.1.

Periodically Cura pass for few milliseconds and it makes possible to click some button and than wait again.

### Expected results

I expected any else result.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7586431/cura.log)

fvrmr, 2021-11-23:
Hi @digiTALker thank you for your report.
It's a bit hard to see in the logs the connection between Autodesk and Cura
Could you try to delete al your folders from previous Cura versions (4.9 till 4.12.1) from this directory `C:\Users\<your username>\AppData\Roaming\cura`  **Don't delete 4.8 since that is working.** 
After you deleted this try to install 4.12.1 again and see if it works. 
Let me know if that works

digiTALker, 2021-11-24:
So I have something to say.
I removed everything. 
And installed 4.12.1. 
First launch was OK. 
After signing was OK.
After 3 plugins updates  was OK.

Than I closed Cura and open again. And my problem starts again.
Better to see in video.
[https://youtu.be/OhtJIHrn4rM](url)

Ghostkeeper, 2021-12-02:
Since Cura 4.9 we've updated to a newer version of Qt, which may have something to do with it.

I don't know how the Autodesk and Google software interfere with Cura though. No other software should really be involved with what Cura does. Software that runs with administrator privileges can interfere (and antivirus software sometimes does interfere with Cura calling CuraEngine.exe). The software you're killing in Task Manager probably runs as administrator. Because the Autodesk software is licensing software meant to prevent the user from meddling with it, and because Google tends to want access everywhere ;)

In the log files it shows that Cura started successfully and then gets abruptly interrupted during (seemingly) idle time. The interrupting is done by the task manager. So there is not really anything that can be read from there about what's causing this.

I know that a lot of people at Ultimaker use Fusion together with Cura without any problems, so unless the licensing software for that is different from what you're using, the problem doesn't seem to occur everywhere. Maybe we can see if any other reports about this come in, to determine if there are any patterns of where the problem occurs.

PHannan80, 2022-01-12:
I am experiencing similar issues.
Over the christmas period, I installed AutoDesk Connector. I was away from my printer so had no reason to open Cura.
On returning home, and wanting to use Cura, it now errors after opening with "Not Responding".

AutoDesk Connector has been uninstalled. OneDrive has been paused and exited. GoogleDrive is closed. No other network drives exist.

Have attempted uninstall/reinstall 4.12.1
Tried 4.13 Beta
Have now installed 4.13

The outcome is the same for each, after opening the application it very quickly falls in to a "Not Responding" state. It periodically gets to a point where it appers to be responding, but any interaction results in a further "Not Responding" state.

Being disconnected from the internet does not help. The same outcome results.
[cura.log](https://github.com/Ultimaker/Cura/files/7854241/cura.log)
.

digiTALker, 2022-01-13:
it`s Not needed to uninstall, just enough kill the process.
Also try to run Cura via CuraCli.exe

PHannan80, 2022-01-13:
Unfortunately, I have tried this. The result is unchanged

PHannan80, 2022-01-13:
> it`s Not needed to uninstall, just enough kill the process. Also try to run Cura via CuraCli.exe  

Unfortunately, I have tried this. The result is unchanged

PHannan80, 2022-01-13:
Update: I have installed Cura 4.9 which appears to operate without a problem. But obviously, missing all of the nice new features

TheKrush, 2022-01-17:
It's frequently using a lot of CPU causing frequent freezing.

sbsoneji, 2022-01-25:
> It's frequently using a lot of CPU causing frequent freezing.

same here but I noticed one thing when it uses more CPU power it does not take advantage of all cores, it fully loads 1 core and 2 threads at a time

sbsoneji, 2022-01-25:
> It's frequently using a lot of CPU causing frequent freezing.

This happens to me every time after installing windows 11.
I have a weird fix but if works every time for me, open 2nd cura window when cura freezes 1st time (don't close 1st window, keep it open minimized or in background) and work in 2nd window.
Keep note that I open it as admin every time and I have checked "Disable full-screen optimizations" in compatibility section in the properties of cura.exe

GregValiant, 2024-11-16:
Is this still a problem in current versions of Cura (5.8.0 and up)?  Can this be closed?

PHannan80, 2024-11-16:
Seems to have been fixed. Can close

closed 2024-11-16 by GregValiant