Ultimaker/Cura #10906: No retraction / z hop during solid infill printing

Matszwe02, 2021-11-23:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

Import any model
the travel lines inside solid infill aren't retracted neither z-hopped, which harms the external surface look

### Actual results

[image]

### Expected results

Cura should avoid scrubbing the solid infill with a nozzle, because it leaves clearly visible lines

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

-

fvrmr, 2021-11-23:
Hi @Matszwe02 thank you for your report. 
The z hop is not visible in the preview mode. If you export your gcode and then import that gcode in Cura it should show the z hops in the preview animation.
Let me know if that works for you and if not could you share a project file?

Matszwe02, 2021-11-23:
> Hi @Matszwe02 thank you for your report. The z hop is not visible in the preview mode. If you export your gcode and then import that gcode in Cura it should show the z hops in the preview animation. Let me know if that works for you and if not could you share a project file?

[image]
[image]
[image]

As you can see, when I view the gcode, some travel moves are with retraction and z raise, but these aren't.
I posted the print images, which have lines visible both on top and bottom. That means that it wasn't raised neither retracted.
It happens on every model, for example: https://www.thingiverse.com/thing:2332471

Matszwe02, 2021-11-23:
During the print I can even see that between printing perimeters it moves over the print bed leaving a line, so it is underextruding for the first mm of printing the next perimeter or infill (due to lack of nozzle pressure)

smartavionics, 2021-11-23:
Looks like you don't have combing enabled with the "no skin" combing setting?

Matszwe02, 2021-11-24:
> Looks like you don't have combing enabled with the "no skin" combing setting?

I tester with all combing modes

Ghostkeeper, 2021-12-02:
If the combing mode is set to "Not in skin", then it's expected to move through the walls. If it's not necessary to cross any walls it can comb, and stay within the model. It doesn't feel the need to retract for those moves because oozing is not visible from the outside.

Of course, oozing indeed still does reduce the pressure in the nozzle chamber. If retractions/z hops are preferred, I suggest you disable combing. It should then decide to retract. Alternatively, you could set a Max Comb Distance With No Retract to retract only for longer moves and still make the combing movements. Make sure that the maximum retraction count is high enough, though, either way.

Let us know if this helps. If not, please provide a project file using File -> Save Project. This file also contains the configuration you're using in your printer, your current settings and profiles. That way we can reproduce the print exactly.

Matszwe02, 2021-12-02:
> If the combing mode is set to "Not in skin", then it's expected to move through the walls. If it's not necessary to cross any walls it can comb, and stay within the model. It doesn't feel the need [...]

Oozing is still visible in the bottom surface (see pictures), but the lack of z hop also affects the top layer, so the nozzle remelts its path through the print. I'll check if your suggested solution works. Thanks for help for now.

Matszwe02, 2021-12-10:
I tested it and without combing it doesn't show any artifacts. But anyway you can consider making the setting that the first and the last layers are without combing (+ retraction) so the surface looks as good as possible. The issue is very visible when I need to print many small holes, but everytime printer jumps from one to another, it loses a bit of filament and doesn't print the entire hole due to lack of pressure.

Ghostkeeper, 2021-12-14:
> But anyway you can consider making the setting that the first and the last layers are without combing (+ retraction) so the surface looks as good as possible.

Indeed, in Cura 4.12 we added the "Not in outer skin" option for combing that does precisely that.

closed 2021-12-14 by Ghostkeeper