Ultimaker/Cura #10907: No funciona la carga de archivos STL arrastrándolos a Cura

matosantos, 2021-11-23:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Anet ET4

### Reproduction steps

Arrastrar STL al área de impresión

### Actual results

No se puede importar archivos arrastrándolos.
Se a probado a ejecutar Cura como administrador sin ningún resultado exitoso.

### Expected results

Al arrastrar un archivo stl al área de impresión de Cura, este debería ser cargado por Cura.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No se si es necesario incluir algún archivo aquí puesto que lo que sucede es algo creo que general y que a todo el mundo le pasará con esta nueva versión.

fvrmr, 2021-11-23:
Hi @matosantos thank you for your report.
According to Google translate you can't import stl files. 
Do you have Chitubox installed if so see: https://github.com/Ultimaker/Cura/issues/9668

And please for next time fill in the template in English, my Spanish is not that good.

matosantos, 2021-11-23:
> Hola @matosantos, gracias por tu informe. Según el traductor de Google, no puede importar archivos stl. ¿Tiene Chitubox instalado? Si es así, consulte: [# 9668](https://github.com/Ultimaker/Cura/issues/9668)
> 
> Y por favor, para la próxima vez llene la plantilla en inglés, mi español no es tan bueno.

I do not have Chitubox installed, I have done a clean installation of Cura removing previous versions in case this solves the problem and the problem is still not solved.

fvrmr, 2021-11-23:
Could you then share your log file? You can find your log file here: `C:\Users\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

matosantos, 2021-11-23:
[cura.log](https://github.com/Ultimaker/Cura/files/7587967/cura.log)

matosantos, 2021-11-23:
> ¿Podrías entonces compartir tu archivo de registro? Puede encontrar su archivo de registro aquí:`C:\Users\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`
[image]

Ghostkeeper, 2021-12-02:
I see in the log that other methods of importing files seem to work correctly. At least, I see an STL file `suplemento para lineal Y Part1.stl` being loaded and moved around on the build plate. There are no entries about loading an STL file by dragging it onto the window.

I'm really not sure what to do here. We can't reproduce this ourselves. The code that handles the dragging is foolishly [in QML](https://github.com/Ultimaker/Cura/blob/95f0b692a646ac88e01fb532b8af8e22dbdcd751/resources/qml/Cura.qml#L667-L670) where we don't get as nice logs or crash reports. One thing that could maybe be done to get more information is to run Cura through a command line: Win+R, type cmd, and in the command line run `"C:\Program Files\Ultimaker Cura 4.12\Cura.exe"`. We're looking for lines starting with `qml:` in the resulting terminal output then, when the drag-and-drop is done in the Cura window.

Other than that, maybe other people can reproduce this too and we'll be able to see the commonalities between you two to debug what could be causing this problem.

matosantos, 2021-12-06:
> Veo en el registro que otros métodos de importación de archivos parecen funcionar correctamente. Al menos, veo un archivo STL que `suplemento para lineal Y Part1.stl`se carga y se mueve en la placa de construcción. No hay entradas sobre cómo cargar un archivo STL arrastrándolo a la ventana.
> 
> Realmente no estoy seguro de qué hacer aquí. No podemos reproducir esto nosotros mismos. El código que maneja el arrastre está tontamente [en QML](https://github.com/Ultimaker/Cura/blob/95f0b692a646ac88e01fb532b8af8e22dbdcd751/resources/qml/Cura.qml#L667-L670) donde no obtenemos registros tan buenos o informes de fallas. Una cosa que quizás se podría hacer para obtener más información es ejecutar Cura a través de una línea de comando: Win + R, escriba cmd y en la línea de comando ejecutar `"C:\Program Files\Ultimaker Cura 4.12\Cura.exe"`. Estamos buscando líneas que comiencen con `qml:`en la salida del terminal resultante, cuando el arrastrar y soltar se haga en la ventana de Cura.
> 
> Aparte de eso, tal vez otras personas puedan reproducir esto también y podremos ver los puntos en común entre ustedes dos para depurar lo que podría estar causando este problema.

To avoid the problem I have directly returned to the previous version of Cura, with the previous version I have no problem loading files by dragging them.

matosantos, 2021-12-07:
> Veo en el registro que otros métodos de importación de archivos parecen funcionar correctamente. Al menos, veo un archivo STL que `suplemento para lineal Y Part1.stl`se carga y se mueve en la placa [...]

I have also noticed that the shortcut buttons for, move, rotate, resize, etc. do not work.

mmcp42, 2022-04-22:
Cura 4.12.1 used to allow drag xx.stl and drop into window
steps to reproduce:
1. open cura 5
2. open folder with .stl files
3. drag xxx.stl into cura window (note "no entry" mouse cursor)
4. release mouse
5. nothing happens

double-click xxx.stl works as expected (opens cura and opens xxx.stl)

file::open xxx.stl works as expected also

just drag & drop no longer allowed

GregValiant, 2024-11-16:
Is this still a problem in current versions of Cura (5.8.0 and up)?  Can this be closed?

mmcp42, 2024-11-16:
Yes, this can be closed
Thanks

GregValiant, 2024-11-16:
Done.

closed 2024-11-16 by GregValiant