Ultimaker/Cura #10908: Per Model Setting modifying Z seam alignment for overlaps triggers unmodifiable 100% infill density

tokamac, 2021-11-23:
### Application Version

4.12.1

### Platform

macOS 12.0.1

### Printer

Prusa i3 MK3S

### Reproduction steps

Hollow model set to 0% infill, 3 perimeters. 
Z Seam Alignment: Shortest.

When "Per Model Settings" is used with a modifier "Modify Settings for overlaps" on some part of the model, with the only settings : "Cutting Mesh" and "Z Seam Alignment: Sharpest Corner"

### Actual results

Then the infill density within the modified part goes to 100%!
And adding the modifier setting "Infill Density: 0.0%" does not change it at all, the infill stays at 100% density there.

### Expected results

The infill density should not be modified from the general settings when a modifier only modifies another unrelated setting on some part of the model, here the Z Seam Alignment.
Plus, the infill density should correctly drop to 0% when specified so for the modifier, instead of staying at 100%.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No need to upload any log file or model as the bug speaks for itself and is easily replicable.

fvrmr, 2021-11-23:
Hi @tokamac thank you for your report.
With your reproduction steps I am not able to reproduce your issue.
[image]

Could you share a project file (file - save project)?

tokamac, 2021-11-23:
You can't reproduce the issue if you specify that you do not want any Top & Bottom shell for the whole model, and that's what you did in the main print settings highlighted in orange a the upper right corner of your screenshot.
Plus, I see your modifier created new perimeters inside your cube, which is also abnormal. Why are these additional internal walls there? They should not be. 

I want my model to have a top and bottom shells, so these should be set in the main print settings, e.g. Top/Bottom Thickness = 0.8 mm. No internal additional walls neither. Just an outer shell, with 3-perimeter walls, top and bottom shell. Nothing else inside. 
Now, when one adds a modifier to such a model, it creates another top & bottom shells WITHIN the structure of the hollow model, which is not normal, see my screenshot below.

[image]

tokamac, 2021-11-23:
I eventually managed to get rid of the additional internal Top/Bottom shells at the interface of the modifier, adding all these settings to the modifier, see newest screenshot below.

[image]

But in the end, the Z-seam alignment modifier doesn't work at all anymore. It is still "shortest" really, and not "Sharpest corner".

I know it because I have a "scar" appearing on another very specific hollow model of mine which appears only when I choose "shorter" for the Z-seam alignement. I cannot use another Z-seam alignment setting on that whole specific model, because those settings create another artefact, some bad under-extrusions due to much longer travel moves, so I must keep "shorter" (or "random" which creates bits on whole surface that I don't want either). 

Plus, it is easily observable unsticking the Shell box for Line Type: the travels are unaffected by the modifier.

Initially I had the idea to modify the Z-seam alignment to "sharpest corner" only in the area where the scar appears on the complicated model (using "shortest" otherwise everywhere else) in order to offset the seam and force it to be stitched along the edge of the model, and not on its face.

tokamac, 2021-11-23:
Below I show how such modifier does not work at all. The big modifier encompasses the upper half of the model.

First, look at the travel paths for the model where the seam is put on the sharpest corner, and the modifier is the same also. Clean paths:
[image]
Now, look at the travel moves for the same model when the seam is put at Random alignments, but the modifier asks for that seam to be at the sharpest corner for the upper part of the model: The modifier does nothing! The travel paths and the seam keeps being random.
[image]
!

smartavionics, 2021-11-23:
> Alas no, GitHub does not allow to attach 3MF files to bug reports and messages. That's ridiculous BTW!

ZIp it up or rename the .3mf to .zip

tokamac, 2021-11-23:
Of course! Here it is (zipped)
[pyramid.zip](https://github.com/Ultimaker/Cura/files/7591139/pyramid.zip)

Ghostkeeper, 2021-12-02:
This is expected behaviour. Not really desired in your case, but expected because the cutting model indeed cuts the model into separate pieces, each of which get separate walls and skin. You're not bothered by the separate walls here (because your model is cut horizontally only) but you are bothered by the separate skin here. The Top Layers and Bottom Layers settings are set to 4 here, so it'll add 4 layers at the top of where the model is cut, and 4 layers at the bottom of the cutting mesh, resulting in 8 layers of boundary. Then a centimetre higher, it'll add another 4 layers at the top of the cutting mesh and 4 layers at where the piece of the real pyramid starts again.

It can be prevented by adding another cutting mesh to change the top/bottom layers for the bottom side of the top of the pyramid, and setting the Top/Bottom Layers to 0 for the cutting mesh. In your case you can even set the Top Layers to 0 because the pyramid should never get any top layers. Here is an example: [pyramid_no_topbottom.zip](https://github.com/Ultimaker/Cura/files/7641360/pyramid_no_topbottom.zip)

I'd say that this is a duplicate of this feature request then, which requests looking at where cutting meshes cut away from the mesh and preventing skin from being generated there: https://github.com/Ultimaker/Cura/issues/9218

closed 2021-12-02 by Ghostkeeper