Ultimaker/Cura #10909: Will not accept 0 for "Initial Layer Height", and, has no setting I can specify to inform Cura of my nozzle-to-bed levelled distance

gitcnd, 2021-11-23:
### Application Version

4.11.0

### Platform

Mac

### Printer

Anycubic Chiron

### Reproduction steps

1. Level your bed with a 0.2mm gap.
2. Try to print a first layer that is 0.2mm high.

### Actual results

1. Blocks me from putting 0 into "initial layer height".
2. If I put 0.2mm in there, I end up with an unwanted 0.4mm height, with all the flow rates wrong for this gap.

The word "Initial Layer Height" is ALWAYS wrong.  Every 3D printer has at least some initial gap between the nozzle and the build plate, so "Initial Layer Height" is really "Additional First-Layer Layer Height" or whathaveyou.

Basically - this is a logic oversight in the core program.

### Expected results

1. My first layer should have Z=0.0
2. There should have been a way for me to tell Cura that my gap is 0.2mm, so Cura can calculate the correct flow rate based on my nozzle size for this distance.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

See https://github.com/Ultimaker/Cura/issues/10893

I initially thought this was a bad documentation issue, but on closer inspection, I find it's actually a bug
[image]
[cura.log](https://github.com/Ultimaker/Cura/files/7587883/cura.log)

gitcnd, 2021-11-23:
Note also that adding "dust" (e.g. 0.01mm) upsets anything (Plugins, gcode processors, etc) that's expecting layer-height multiples (e.g. temperature test towers etc)... plus I think that adversely affects the initial layer flow rates resulting in no adhesion (I only tested this once, and I switched to another slicer when my attempt with Cura failed).

5axes, 2021-11-23:
Use an other software is a good solution as Cura is full of bugs. You cannot set to Zero also layer height, Line Width , flow Rate etc ... This is not worthy of a professional solution  😄 
[image]

GregValiant, 2021-11-23:
I'm not getting it here folks.  How would you use a Zero layer height?  Just turn the machine off?  There is much mathematics going on in the background and dividing by zero will still cause an error just as it did when y'all were in grade school.  If  I have a 1mm thick model and divide it by 0 then even if that was legal I would have an infinite number of layers.  Additionally, since the layer height is zero then the flow is zero.  A zero line width results in the same problems.  I just don't get it.

@5axes  - I've been using Cura for a couple of years on a variety of different types of models and I have not found it to be "full of bugs".  It isn't perfect and I'm sure that the Cura team (who from afar appear to be very bright, very dedicated people) is constantly working to clean them out.  The program has to run on Linux, Windows, Macintosh, and I think there is a version for Android AND over 300 different printer models are supported (of which only a few are Ultimaker products).  In addition it has to run on a variety of processors.  As each of those operating systems develop and change then Cura must react to keep up.  Frankly I don't know how they manage it.  If you have a bug report then this is the place to report it.  They are taken seriously but there are finite resources to apply and so the bug reports (and feature requests) appear to be prioritized someway by the Cura team.  Ultimaker provides this software for free to everyone.  If you don't care for it then there are a lot of options.  I currently have 16 slicers installed on my computer.  Some are better than others and a couple really suck but for the most part they are passable.  At the top of the pile are PrusaSlicer and Cura.  I like IdeaMaker too.  They are all fairly complicated pieces of software and too many people look for the "Easy" button.  Well, there ain't one.  

@gitcnd - The end of the nozzle is Z = 0.  When we level a printer we are bringing the build surface to the nozzle so that the top of the build surface is Z = 0.  If you set a 0.2mm initial layer height within Cura then your first layer should be .2mm thick.  If you are setting the initial "nozzle to plate" gap at 0.2 then you will indeed get a .4 layer and that .2 will stay throughout the print.  I don't understand why you would do that.  I've certainly never heard of anyone else doing it.  I use .04mm parchment paper for leveling and it works well for me.  A store receipt would be acceptable but nothing sticks to parchment paper.

gitcnd, 2021-11-24:
@GregValiant - yeah - you really don't get it.  Z=0 is NOT the point where the nozzle touched the plate - that NEVER happens (did you read the manual you got with your printer?).  There is ALWAYS some kind of gap - otherwise the nozzle would destroy itself and the build plate when you hit "home".

"Divide by zero" is just you not getting it.  That never happens OF COURSE, because Cura is supposed to understand what the Z=0.0 clearance is supposed to be (usually 0.1mm, the thickness of the levelling sheet you got with your printer).

And unfortunately yes - Cura is totally riddled with bugs.  Drives me crazy *every* time I try to use it - printing on thin air, bizarre movements adjacent single-wall extrusions, broken algorithms for infill that treble the travel movements needed per layer, and all the others that I've raised bug reports about.

This is "open source" - it's written by unpaid non-professionals in their spare time.  Us reporting issues is how they're supposed to get fixed, although most of the time it looks like devs prefer to argue about mistakes, than fix them :-( ... one of the side-effects of low-skilled free labor...

Asterchades, 2021-11-24:
For those who clearly didn't pick up on it, @5axes' post was pretty much 100% sarcasm. His point is that settings which won't work if set to 0 aren't allowed to be set to 0, even in other slicing software.

The Initial Layer Height is not an offset - it's the actual thickness it will attempt to print the initial layer. If you set 0 for this you're basically telling it you don't want it to print the initial layer, which obviously isn't going to work.

Z=0 _is_ the point where the nozzle is touching on the bed. That's exactly what it's supposed to be and that's what it's being treated as. Or, more exactly, Z=0 is the point where you've told your Z axis to be at its minimum, so if it is not touching on the bed it's because you've set the offset wrong yourself - such as using the "sheet of paper" method - or deliberately imposed a slightly larger (or smaller) gap. You should never actually perform any operations at Z0, however, as this will (rather expectedly) drag the nozzle across the bed.

If your initial layer height is 0.2mm, the very first layer should go down at Z0.2. If your printer is moving to Z0.4 it's because you've got a 0.2mm offset configured (Cura cannot do this by default and requires a plugin to set a Z offset, where some firmwares also allow such options). If its going to Z0.2 but you're measuring 0.4mm it's because your Z0 point is 0.2mm higher than it should be (or because your Z steps are twice what they should be).

Note that this behaviour is common across every single free slicer currently available and still in active development. It's also standard in the CNC router world. There is no precedent for it being otherwise and, as such, it should not be expected to be otherwise.

5axes, 2021-11-24:
@Asterchades  thanks a lot and concerning my post you can remove "pretty much" . But it's also because this thread is just the avatar of this one : https://github.com/Ultimaker/Cura/issues/10893

Asterchades, 2021-11-24:
Ah! I thought the complaint looked familiar. Didn't realise just _how_ familiar it was.

5axes, 2021-11-24:
Love also so much one of his previous cura "issue" https://github.com/Ultimaker/Cura/issues/9126 😄

nallath, 2021-11-24:
> This is "open source" - it's written by unpaid non-professionals in their spare time. Us reporting issues is how they're supposed to get fixed, although most of the time it looks like devs prefer to argue about mistakes, than fix them :-( ... one of the side-effects of low-skilled free labor...

I warned you the last time you decided to insult the developers. I actually warned you _twice_ already. Goodbye!

closed 2021-11-24 by nallath