Ultimaker/Cura #10910: Cura 2.3.1 "fatal exception"

albi53, 2021-11-23:
### Application Version

2.3.1

### Platform

Windows 10

### Printer

Prusa Mendel I3

### Reproduction steps

For months I did not get problems, since a couple of days when I launch Cura and get the warning
"Fatal exception occurred...."
Use the information below to post e bug report
I'm not able to track any chnges in my PC to justify the change in behaviour

### Actual results

I'm posting the bug report

### Expected results

Hopefully Cura 2.3.1 is able to function again
----------------
Version: 2.3.1
Platform: Windows-10-10.0.19041
Qt: 5.5.1
PyQt: 5.5.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 61, in <module>
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\Application.py", line 310, in getInstance
  File "D:\2.3\build\inst\lib\python3.4\site-packages\cura\CuraApplication.py", line 173, in __init__
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\Qt\QtApplication.py", line 94, in __init__
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 107, in upgrade
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 281, in _upgradeFile
MemoryError

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Any other information, presently I'm usign Cura 15.4.6 without problems, woud like to return to the newer version

fieldOfView, 2021-11-23:
Deleting cura.log from C:\Users<your username>\AppData\Local\cura should fix this.

Note that your version of Cura is nt officially supported. You may need to upgrade at some point.

fieldOfView, 2021-11-23:
Duplicate of https://github.com/Ultimaker/Cura/issues/1312 which has long since been fixed in newer versions of Cura.

closed 2021-11-24 by nallath