Ultimaker/Cura #10912: Errors and more errors

NunoFilipePereira, 2021-11-23:
### Application Version

4.12.1

### Platform

Mac

### Printer

Bíqu b1/ tarantola

### Reproduction steps

Errors and more errors
[image]

### Actual results

Errors and more errors

### Expected results

Errors and more errors
[cura.log](https://github.com/Ultimaker/Cura/files/7589390/cura.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7589395/cura.log
[image]
[image]
[cura.log](https://github.com/Ultimaker/Cura/files/7589421/cura.log)
)

GregValiant, 2021-11-26:
867 errors by Washington Nationals in 1886 (in 122 games).
1. Herman Long with 122 in 1889
1. Billy Shindle with122 in 1890

fvrmr, 2021-11-30:
Hi @NunoFilipePereira thank you for your report.
In your object list you have custom support added. Could you delete those? 
Also could you share your project file (file - save project)?

closed 2021-12-14 by no-response[bot]