Ultimaker/Cura #10913: Add Thumbnail to 3MF output format

pedrolamas, 2021-11-23:
### Is your feature request related to a problem?

3MF format supports including a thumbnail, so it would be great if the files created by Cura would include that thumbnail that can be easily read in Windows file explorer (and possibly other OS) 

### Describe the solution you'd like

Add a thumbnail of the model to the 3MF output file

### Describe alternatives you've considered

Adding the GCode postprocessing script to add thumbnails, however those are not visible in Windows file explorer, only on specific files.

### Affected users and/or printers

Every Windows user will benefit from this (and possibly other users too)

fvrmr, 2021-11-30:
Hi @pedrolamas thank you for your feature request.
I have discussed it with the team and we decided to create a ticket for this on our backlog.
Devs see CURA-8723

pedrolamas, 2021-11-30:
Hi @fvrmr, Sorry if that is not the correct process but I just went ahead and created this as a Feature Request (FR) type of ticket without discussing it with anyone before.

fvrmr, 2021-11-30:
It is the right way to request a new feature for Cura. I brought it up with the Cura team and based on your request we have now a ticket on our backlog for this :)

nallath, 2021-11-30:
I'm personally a big fan of this feature! It shouldn't be too hard to implement since we already have a system that can create these thumbnails.

fieldOfView, 2021-11-30:
See #10971.

PR #10971 (2021-12-02): Add a thumbnail to 3mf files (projects & exports)
This PR adds a thumbnail to project files (edit: and exported files), like they are added to UFP files.

This PR fixes #10913
closed 2021-12-02 by nallath