Ultimaker/Cura #10914: Allow plugin DisplayProgressInLCD to use M118 besides M117 and M73

inigoml, 2021-11-23:
### Is your feature request related to a problem?

I've an Artillery printer with BigTree Tech TFT. This display does not show progress when printing via Octoprint due to not manage M117 messages. It would be nice to add M118 serial message support in order to allow send messages to TFT.

### Describe the solution you'd like

Add not only M73 or M117 messages but also M118.

### Describe alternatives you've considered

Print form TF or USB without using Octoprint.

### Affected users and/or printers

Artillery Sidewinder X1/X2, Genius, Genius Pro.

### Additional information & file uploads

[DisplayProgressOnLCD.zip](https://github.com/Ultimaker/Cura/files/7590545/DisplayProgressOnLCD.zip)
Attached modified script for Cura 4.11.2. Easy to integrate in Cura main branch.

inigoml, 2021-11-23:
This is how it's seen from Cura.
[image]

rust84, 2021-11-28:
Nice feature. I would suggest P0 as a default rather than P2 and this will broadcast it on all serial ports for people who have a display connected to P1. Tested with an Ender 3 and SKR Mini E3 V2 + TFT35. 

Would be useful with https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/2241

digant73, 2021-11-29:
it should be better that the current plugin can be enhanced to support the commands in PR bigtreetech/BIGTREETECH-TouchScreenFirmware#2241 (now merged on the last BTT TFT fw). In particular, the plugin must support also actions:
action:print_start
action:print_end
allowing the TFT to switch/close the Printing menu

fvrmr, 2021-11-30:
Hi @inigoml thank you for your request. Since you've already made a script you can create a pull request! Let me know if you have questions about creating a pull request!

inigoml, 2021-11-30:
Updated script for M118 support with @digant73 last TFT mod.
[DisplayProgressOnLCD.zip](https://github.com/Ultimaker/Cura/files/7627808/DisplayProgressOnLCD.zip)

inigoml, 2021-11-30:
> Hi @inigoml thank you for your request. Since you've already made a script you can create a pull request! Let me know if you have questions about creating a pull request!

Will do!

inigoml, 2021-11-30:
https://github.com/Ultimaker/Cura/pull/10972

inigoml, 2021-11-30:
[image]

digant73, 2021-11-30:
@inigoml Just to make clear the feature you are implementing, please add in the description of the PR also a reference to the Octoprint support implemented on the BTT TFT (so https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/2241)

inigoml, 2021-11-30:
@digant73  Done!

closed 2021-11-30 by inigoml