Ultimaker/Cura #10916: Sandbox Settings Option for Shared Systems

zxc8027, 2021-11-24:
### Is your feature request related to a problem?

At [The Construct @ RIT](https://hack.rit.edu/), we use Cura on shared, public accounts on desktops. In order to ensure the settings are the defaults that we recommend to users, and the first printer they open with is the same, we attempt to “sandbox” the settings such that they don’t persist between sessions. The use cases of this feature include:
* If a user changes the selected printer, it does not persist to the next user. This reduces user error where the user expects the same default printer every time, and this not happening results in the wrong printer being sliced for.
* If a user changes the selected material, it does not persist to the next user. For the rare cases we use non-PLA filament, this reduces print failures due to loading the wrong filament settings for users.
* If a user changes printer settings, it does not persist to the next user. Some users change settings in ways that make prints more likely to fail or take a lot more time, and this reduces errors for the next user.

### Describe the solution you'd like

From our experience, the solution requires the following requirements:
* This “sandboxing” of settings must either have a way of configuring the sandboxed state or updating the state so we can modify it. This includes actions like updating printers or changing defaults.
* The settings either need to not be saved or be overwritten on load. The exact implementation doesn’t matter as long as the default printer and printer/material settings reset when someone opens Cura. The first option would result in less disk I/O and would prevent issues from 2 Cura sessions being open.

### Describe alternatives you've considered

We’ve gotten around this by modifying the Uranium library sources to point to a different location for the settings (this is our main limitation for why we can’t make a plugin), and disabling calls to save the settings. The specific code changes we do currently aren’t public but can be made public on request if it helps our use case, and potentially other makerspaces looking to use Cura. We can also help with the related code changes if some guidance on what to change can be given.

### Affected users and/or printers

This feature request affects a combination of the users (students, faculty, and alumni) to reduce user confusion and errors, as well as to Lab Managers for maintaining the systems and helping users.

fvrmr, 2021-12-03:
Hi @zxc8027 thank you for another feature request.
I also discussed this request with a developer and also this is not something we want to implement.
It doesn't have any priority or value for us. 

You could make a plugin for this since plugins will be load before the machine and settings are loaded.

I will also close this issue, I hope you understand.

closed 2021-12-03 by fvrmr

zxc8027, 2021-12-08:
> You could make a plugin for this since plugins will be load before the machine and settings are loaded.

Thanks for mentioning this. I tried this out, and this works for our use case with no modifications required in Cura. I'll try to remember to edit this post later once the plugin code is on GitHub in case anyone is searching for this later.