Ultimaker/Cura #10917: Intercept/Block Exporting Prints in Plugins

zxc8027, 2021-11-24:
### Is your feature request related to a problem?

At [The Construct @ RIT](https://hack.rit.edu/), RIT’s makerspace, we have been using a modified version of Cura 4 for over a year to allow users to slice their prints for our 3D printers while providing some control over what they can do and logging what they print to be billed later. In order to log prints, we must have users enter their information before they can save a GCode or X3G file. We have many use cases for this:
* When a user enters their information, we store the name of their file, the material used, and the weight of their print according to Cura. This allows us to bill them at the end of the semester for the prints they owe. We have two reasons to do this 1) to get back some of the money spent on material and 2) to discourage people from printing large amounts of random things for free.
* With the information stored,  we can figure out patterns of who uses prints and for what types of projects. It also helps us predict filament usage for purchasing and printers to buy.
* If a print causes an issue with the printer, we are able to contact the person who printed to figure out what happened and prevent future failures.
* If a user refuses to fill out the form, the print is not saved and the user can not continue.

### Describe the solution you'd like

For our requirements to make a plugin for payment screens and other systems, we need a way to be able to intercept a request to export a print and potentially either delay it by an unknown period of time or block if the user fails to enter their information or another check fails. We do need to know the exact file name though, as well as the weight and material used for the print. Other details, like the printer and the output file, may be needed for others, but we do not have this use case.

### Describe alternatives you've considered

We’ve gotten around this by wrapping the `LocalFileOutputDevice` and `RemovableDriveOutputDevice` to call an external method, which may actually call to write the file in the future. These modifications have made upgrading versions of Cura difficult and making the payment screen a plugin difficult. The specific code changes we do currently aren’t public but can be made public on request if it helps our use case, and potentially other makerspaces looking to use Cura. We can also help with the related code changes if some guidance on what to change can be given.

### Affected users and/or printers

This feature request affects a combination of the users (students, faculty, and alumni) to reduce user confusion and errors, as well as to Lab Managers for maintaining the systems and helping users.

nallath, 2021-11-24:
These are the interesting questions that I love to be able to think about :)

There is an "onWriteStarted" signal that you could hook into. But it's not really intended to force people to actually do something interface wise. It also doesn't allow for any kind of exit (eg: Something isn't the case, cancel the write all together).

If you want to change the g-code (eg; add the extra info that you need for tracking) I would suggest that you add a bit of code that listens to "Application.getInstance().getOutputDeviceManager().writeStarted". This is also what the PostProcessing plugin does. 

I think it would be possible to make the changes that you want, but as interesting as it is, it probably won't get a lot of priority (we unfortunately have a lot of things that we want / need to do, so we have to make tough choices). But I can offer you a suggestion on how you could achieve this in a more stable fashion!

What you could do is move the logging _out_ of Cura to a centralized location. So instead of tracking permission in Cura, you just add an extra output device to Cura (much like the digital factory / UM3 networking plugin does) and send the jobs there. This mini server can then have a small queue, track billing, etc. From within Cura you can then just have a small interface that connects with that server. The server would then be able to send it to the various printers (eg; for instance with something like Octoprint?).

zxc8027, 2021-11-25:
> There is an "onWriteStarted" signal that you could hook into. But it's not really intended to force people to actually do something interface wise. It also doesn't allow for any kind of exit (eg: Something isn't the case, cancel the write all together).

I saw this event before. I don't think putting something blocking or something that throws an exception would be a good idea, and having something else that can block it would work better. What would be better? Don't know. We can handle any changes to the plugins for this, but I don't expect it to happen during the remainder of the school year, and  I would prefer more discussion on a design that isn't difficult to support in the future, or blocks future features.

> What you could do is move the logging out of Cura to a centralized location. So instead of tracking permission in Cura, you just add an extra output device to Cura (much like the digital factory / [...]

We've debated networking the printers before, but ran into 2 issues:
1. With our setup, networking the printers would be hell to put it bluntly. The hardware to do so would be a lot, but the network management at RIT would be much harder to pull off.
2. The lab director (Michael Buffalin IV) wants to encourage people to come into the space and learn about using the hardware, and us having any sort of networking of the printers discourages this. If we were to move to having a networked system in the lab, there will be a lot of people wanting to run Cura out of the lab.

fvrmr, 2021-12-03:
Hi @zxc8027 again thank you for your feature request.
I used to work at a library where I used Octoprint to have access to the printers. You could set up a local network and use passwords per printer. These passwords can be changed every week. 

Also this is again not something we want to implement since it has no priority for us. 
I will also close this issue, I hope you understand.

closed 2021-12-03 by fvrmr

zxc8027, 2022-01-30:
I did some thinking about this regarding options. The simplest way this use case can be addressed for us is just by splitting the `requestWrite` method in the output plugins. For example, in `RemovableDriveOutputDevice`, the `requestWrite` method would go from:
```python
def requestWrite(self, ...):
    file_name = os.path.join(self.getId(), file_name + extension)
    ...
    try:
        Logger.log("d", "Writing to %s", file_name)
    ...
```

To something like the following:
```python
def requestWrite(self, ...):
    file_name = os.path.join(self.getId(), file_name + extension)
    self._performWrite(....)

def _performWrite(self, ...):
    try:
        Logger.log("d", "Writing to %s", file_name)
    ...
```

This would allow us to wrap the theoretical `_performWrite` with the functionality we need without changing the behavior or classes. I will be trying this out for The Construct @ RIT before creating a PR to make sure this works, but if this solution, would a change like this be considered? We *could* write a plugin that patches the plugin sources to do it, and it is how I will most likely test the concept, but I would prefer to not have to do that long-term.