Ultimaker/Cura #10918: Intercept/Block Changing Printers and Materials in Plugins

zxc8027, 2021-11-24:
### Is your feature request related to a problem?

At [The Construct @ RIT](https://hack.rit.edu/), RIT’s makerspace, we have been using a modified version of Cura 4 for over a year to allow users to slice their prints for our 3D printers while providing some control over what they can do and logging what they print to be billed later. In order to limit what users can do with respect to changing printers and materials, we have added authentication requirements for changing these for specific settings. When a user changes a printer or material, it will either change as expected for a printer/material that is allowed or ask for authorization (user to swipe their university id), which will halt and potentially abort changing the printer/material. We do this for a few reasons:
* We sometimes have to validate printers or have specialty printers that lab managers can use but regular users can’t. For example, we had a printer that was relegated to TPU before being retired.
* We only offer PLA to students unless we or students bring in their own filament. To protect the machines that can’t run hotter filaments, we must make sure users can’t freely select materials like ABS while still allowing lab managers to change it for special cases.

### Describe the solution you'd like

For our requirements to make a plugin for controlling changes to printers and materials, we need a way to intercept a request to change these and potentially either delay it by an unknown period of time or block it.

A potential implementation-level solution would be to allow some sort of “pipeline” or series of checks that can be added to by plugins, which can halt or fail. Our system only requires one for a user to authenticate, but more could potentially be used. For example, another makerspace may want to add a confirmation saying that changing the material may cause damage if the printer uses a different material.

### Describe alternatives you've considered

We’ve gotten around this by wrapping the `MachineManager` class and reference in `ConstructMachineManager` to call an external method, which will either just call the original `setActiveMachine` or `setMaterial` method if the printer or material does not require authorization, or open a prompt that may call the original method in the future. The specific code changes we do currently aren’t public but can be made public on request if it helps our use case, and potentially other makerspaces looking to use Cura. We can also help with the related code changes if some guidance on what to change can be given.

### Affected users and/or printers

This feature request affects a combination of the users (students, faculty, and alumni) to reduce user confusion and errors, as well as to Lab Managers for maintaining the systems and helping users.

Ghostkeeper, 2021-12-03:
I think that if we build a system to check permissions for something like that, we'd not implement it client-side in Cura. It's not the best practice security-wise. We can't control what people do with their Cura installations. You might, within the confines of your venue, but we can't. And we are not likely to ever spend time on building such things for you. Ultimaker has, in a way, already developed it through their Digital Factory system, and in a way that is more secure given the attack vector that people might edit files in their computers.

So I think this should be considered out of scope for Cura. We are not going to try to build printer access management in the client. Yes, we could build a partial solution for modders to hook into, but there is really little incentive for us to add that to Cura.

Perhaps an easier way to implement this would've been to subscribe to the `globalContainerStackChanged` or `globalContainerChanged` signals to see when the printers change, and change it back if permissions are not (yet) given. But your solution would work too I suppose.

fvrmr, 2021-12-03:
Hi @zxc8027 thank you again for your feature request.
Like @Ghostkeeper just explained this is not a priority for us to implement. 

I will also close this issue, hope you understand.

closed 2021-12-03 by fvrmr

zxc8027, 2021-12-07:
> I think that if we build a system to check permissions for something like that, we'd not implement it client-side in Cura. It's not the best practice security-wise. We can't control what people do with their Cura installations. You might, within the confines of your venue, but we can't.

We have had people abuse our systems before, and we don't have another option without networking all of our printers. Primarily, the goal of this isn't to stop malicious actors: it is to stop users from using printers we are validating or aren't intended for public use due to various reasons.

> Perhaps an easier way to implement this would've been to subscribe to the `globalContainerStackChanged` or `globalContainerChanged` signals to see when the printers change, and change it back if permissions are not (yet) given. But your solution would work too I suppose.

Listening for `globalContainerChanged` for machine changes and `activeMaterialChanged` for material changes appears to work in a plugin with no invasive changes or patching like before. Thanks for mentioning this. I'll try to remember to edit this post later once the plugin code is on GitHub in case anyone is searching for this later.