Ultimaker/Cura #10921: CURA inserting +20 C temperature M104 code

KD7UM, 2021-11-25:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Ender 3 V2

### Reproduction steps

Sliced the model, saved the gcode, opened OctoPrint, uploaded the gcode, started the print.  The sliced PLA nozzle temperature was either 200C or 205 C.  After the purge stripe is printed the nozzle temperature jumps to 225C on both gcode sets.  I have to manually dial the temp back to 200 or 205C for the print job.

Here is the snip of gcode that is causing the problem:
;LAYER_COUNT:41
;LAYER:0
;[CAZD:
M104 S225.0 T0
;:CAZD]
M107
M204 S500
M205 X8 Y8
G0 F7500 X63.872 Y62.095 Z0.2
;TYPE:SKIRT

Is there a setting in CURA to turn this off or is this a bug.  What is CADZ that the M104 command is bracketed by?
Thanks

### Actual results

The temperature jumps to 225C that requires a manual reset to 200C or 205C

### Expected results

The temperature should stay at the sliced temperature not increasing to 225C
[cura.log](https://github.com/Ultimaker/Cura/files/7599566/cura.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[CE3_Merry_Christmas_reduced_size-new.gcode.txt](https://github.com/Ultimaker/Cura/files/7599704/CE3_Merry_Christmas_reduced_size-new.gcode.txt)

GregValiant, 2021-11-25:
Immediately to the left of the Slice button is an icon for post-processors.  Yours will have a red circle on it indicating that a post-processor is active.  In your case it is "Change At Z".  Click on the icon and in the following dialog you will see 3 little icons next to Change At Z and one of them is an "X" to dismiss the post-processor.
There are several post-processors available and you should probably get to know them enough to know what they do.  On the menu bar select "Extensions | Post-Processing | Modify Gcode" to get to them.  In particular I find Pause at Height handy dandy for changing colors during a print.  I have had up to 7 Change At Heights enabled for one multi-color print.  Using "by height" doesn't work if there are z-hops in a file, but "by layer" works very well so long as the firmware on your V2 supports the M0 pause command.  If it doesn't, then there is a workaround.

[image]

KD7UM, 2021-11-25:
Awesome, thanks for the quick response.  I built a series of temperature changes to determine the best nozzle temp for a new PLA and they were built as custom post processors.  Somehow it became the default in CURA for all future projects.  I haven't done any color changes but I recently moved from Creality firmware to Jyers UI although the most current release has some issues so I reverted back.

closed 2021-11-25 by KD7UM

GregValiant, 2021-11-25:
Once you turn one on it stays on until you turn it off.  I've had prints pause because I forgot to turn off Pause at Height.  I come back expecting it to be done and there is sits, waiting patiently for the dummy to restart it.