Ultimaker/Cura #10924: Thermal Runaway and/or Heating Failed when printing hot materials on 4.11 or 4.12

jonathanalevi, 2021-11-25:
### Application Version

4.12

### Platform

Mac OS X

### Printer

Geeetech A10M

### Reproduction steps

1. start a print with any material that requires high heat (235+)

### Actual results

Printer will either tell me "Thermal runaway" or "heating failed" - usually during the first layer. 

**Please note that this only happens on 4.11 or 4.12. I contacted the manufacturer, and they told me to revert to Cura 4.10, and that seems to have fixed the issue**

### Expected results

The print should work fine on 4.11 or 4.12 as well. 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7603025/cura.log
[100 spacers with 0.8.zip](https://github.com/Ultimaker/Cura/files/7603027/100.spacers.with.0.8.zip)
)

GregValiant, 2021-11-26:
I opened the 3mf file in Cura 4.12.1.  Slicing with the posters settings - The M104 / M109 commands in the gcode file are always 250° no matter what the hot end temperature settings are in Cura.

The StartUp G-Code looks correct and a replacement is being made.
M140 S{material_bed_temperature_layer_0} ;set bed temp
M104 S{material_print_temperature_layer_0} ;set extruder temp and wait
G28 ; home
G29 ;Auto Bed Leveling (BlTouch)
G90 ;absolute positioning
M109 S{material_print_temperature_layer_0} ;set extruder temp and wait

But within Cura, no matter what I enter as "Printing Temperature Initial Layer" the results are always...
M104 **S250** ;set extruder temp and wait
G28 ; home
G29 ;Auto Bed Leveling (BlTouch)
G90 ;absolute positioning
M109 **S250** ;set extruder temp and wait

So maybe that 250° is pushing the printer past it's max temp?  Again, no matter what I enter for Print Temp and Initial layer print temp the gcode comes out as M104 S250.

Switching printers from the posters "A10M Custom" to a standard definition Geeetech A10M did NOT produce the same result as the temperatures were as they should be.

jonathanalevi, 2021-11-27:
Any idea why it's always setting at 250? I HAVE noticed that it always seems to force the print at 250...?
Is this a bug, or something I did wrong in my start g-code? 

Thanks for taking the time to check into this for me.

jonathanalevi, 2021-11-27:
BTW I've had thermal runaway / heating failed issues before I changed my start G-Code, as well. The annoying thing is, sometimes it will print fine for 18 hours, finishing prints. Other times, it fails on the first layer.

GregValiant, 2021-11-27:
It appears that because the brim is set for Extruder 2 that the temps for that extruder and Spider ASA are what we are seeing as Extruder 2 is what starts out the print.  If you will set the brim to Extruder 1 it appears that all is well in the Gcode.  

As far as the temperature runaway goes, that's between you and your printer I think.  Since Extruder 2 is set for the "Spider ASA" with a 250° printing temperature then that is possibly too high for your printer.  You could try 248° and see how it goes.

fvrmr, 2021-12-03:
Hi @jonathanalevi thank you for your report.
I am not able to reproduce this issue. Is it possible that you changed the settings for extruder 1 but wanted to print with extruder 2?
And about the thermal runaway I agree with @GregValiant that this is not something related to Cura.

closed 2021-12-17 by no-response[bot]