Ultimaker/Cura #10926: Preview mode- no Speed color scheme, only Line type

djordjedr, 2021-11-25:
### Application Version

4.11.10

### Platform

Linux Ubuntu

### Printer

Creality Ender 3

### Reproduction steps

1. Slice your model
2. Go to preview 
3. Select Color Scheme-layer lines
 

### Actual results

Note that user cant switch from Color scheme-line type

### Expected results

Multiple options should be possible for Color scheme 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fieldOfView, 2021-11-25:
Your preview is in "compatibility mode" because either your GPU or the driver you use does not support OpenGL 4.1, which is required for the full featured layer view.

fvrmr, 2021-11-30:
Hi @djordjedr thank you for your report.
Like @fieldOfView mentioned this is because you are in compatibility mode. 
I will close this issue, since this is something we won't fix.

fvrmr, 2021-11-30:
Also if in the preferences the `Force layer view compatibility mode` is enabled you could also disable this. But if it is disabled and you still have compatibility mode this is related to your GPU or driver not supporting OpenGL 4.1

closed 2021-11-30 by fvrmr

djordjedr, 2021-12-01:
Thank you for clearing up this misunderstanding in such a short notice and
sorry for taking up your time