Ultimaker/Cura #10927: wrong widget size for start/end gcode in printer config

petaflot, 2021-11-26:
### Application Version

4.9.1

### Platform

Gentoo GNU/Linux

### Printer

creality 10S Pro

### Reproduction steps

1. emerge cura and its dependencies
2. launch cura for the first time
3. select printer when prompted

### Actual results

https://www.imgpaste.net/image/KJqctY

### Expected results

I expect the content of the G-code widgets to be actually readable and editable ; instead, they are not even large enough for a single line of text.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

nothing I can think of

fvrmr, 2021-11-30:
Hi @petaflot thank you for your report. 
Are you using the official AppImage of Cura?

petaflot, 2021-11-30:
with appimage it works.

I was using 4.9.1 on gentoo

Ghostkeeper, 2021-12-06:
The AppImage is the only version we have the manpower for to support. I reckon that the version on Gentoo Packages is not setting the same environment flags for Qt as we are in the AppImage, which is understandable but can cause differences like this. Changing the environment flags needs changing other things as well. Or perhaps they are using different versions of dependencies.

We'll close this because it works correctly with the AppImage distribution.

closed 2021-12-06 by Ghostkeeper