Ultimaker/Cura #10928: Crash at startup Cura 4.11 and others...

RyzenSavior, 2021-11-26:
### Application Version

4.11.0

### Platform

windows 10 pro and 11 pro Rx 5700 and Rx 6800 xt.  cpu Ryzen 3600, 3700x, 5800x

### Printer

custom (TenLog D3 pro)

### Reproduction steps

Ran program. Cura crashes towards the end of module loads or shortly thereafter. if after screen is usually white but will sometime flash the actual GUI before crashing. Have tried on all the above systems and every configuration of those parts also with cura 4.9 all on x470 and x570 boards. some crash somewhat less frequently... but all configuration's crash >70% and all the way up to 95% of the time. the strange thing is that occasionally it will work. 

I think it would be good to mention that my network configuration is somewhat unusual. As I've experience similar startup like issues that appeared to be random but ended up due to firewall issues (i use Kaspersky) or transfer protocol/ encryption issues particularly with programs that required some sort of authorization from their parent sites. (I did notice cura hangs if not provided a internet connection. Any ways I live in the boonies so my internet is made up of 2 5g phones connected to a server via pda net and easytether the channel bonded using speedify vpn before being sent to a router and distributed to other devices. I have tried all configurations of this setup. vpn on or off, 1 phone 2 phones, device attach to host running cura etc... nothing in this case. Please help I bought simplify because I was frustrated with cura 4.9 but I like cura so much more... I just hate spending 30mins to try and find a sweet spot where it will run
[cura.log](https://github.com/Ultimaker/Cura/files/7606681/cura.log)
n.

### Actual results

crashes before start most of the time

### Expected results

load program, allow use.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7606688/cura.log)

fieldOfView, 2021-11-28:
Do you have multiple displays? If so, in the General preferences there's an option "Restore window position on start". If you uncheck that, Cura will always start on the primary screen. This may fix your problem.

RyzenSavior, 2021-11-28:
I do have multiple monitors, I tried checking that option but to no effect. It still takes about 10 tries before Cura opens and stays opened without crashing. Thank you for the suggestion though.

fieldOfView, 2021-11-28:
The idea was you *uncheck* it; it is checked by default.

RyzenSavior, 2021-11-28:
Yes I understood that sorry if I misstyped. I should have said I tried that
instead of I checked that. What I meant was that I looked at it and tried
your suggestion and that it did not help.

RyzenSavior, 2021-12-03:
anyone with other ideas? or is anyone else experiencing a simular issue with cura? I though for sure there would be more people out there... heck the only two people I know with 3d printers (one also with a tl-d3 pro the other with an ender 3 v2) also have this problem most of the time and have since off and on since cura 4.7 At that rate I would expect many others to have simular issues. Is it an AMD thing? Sure not CPU based any more since ryzen, but is this just compiled to use amd GPU drivers or something? although it does act like a server authorization problem... its a shame I literally had to pay $150 for simplify 3d because cura wouldnt work reliably...

fvrmr, 2021-12-03:
Hi @RyzenSavior you could also try to run Cura as administrator.
And did you try 4.12.1?

RyzenSavior, 2021-12-03:
I am currently running 4.12.1 and have it in administrator. It takes an
average of 10 failed loads before it will stay and not crash.

Ghostkeeper, 2021-12-06:
I'd say that running Cura as administrator is not a good idea, since it should require nothing that needs administrator permissions. Once ran as administrator, any configuration files Cura creates will be owned by the administrator too and can't be modified any more when Cura is not ran as administrator any more in the future. So it can cause issues down the line.

The log file sadly says very little. It's not a crash in Cura's code, because then you'd see a normal crash window and we would see a traceback in the log. The log contains no traceback. We can only see that it crashes just after loading the shaders, which is when it creates the main window. Creating the main window executes a lot of QML code. I'm supposing the error is somewhere in Qt. Where in our interface the crash is created is unknown though.

Perhaps we can wait and see if anyone else reports the same symptoms, so that we can compare what the commonalities are between your computer and others.

RyzenSavior, 2021-12-07:
thank you for the reply. I will do a fresh install to remove admin privilages then. And continue to work with it and see if there are any changes. thank you.

Kuxer, 2021-12-21:
Hello! I have the same problem. I also try ewerything (reinstall different versions of Cura, update nvidia driver and so on). The last line in the log is - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [60]: Loading [C:\Program Files\Ultimaker Cura 4.12.1\resources\shaders\xray.shader]... Full log included.
[cura.log](https://github.com/Ultimaker/Cura/files/7757225/cura.log)

RyzenSavior, 2021-12-21:
Strangely I have found great improvement in symptoms so long as I don't maximize cura window in windows. I can maximize it to full screen while using cura, but so long as I switch it to a windowed mode even just 90% of full screen before exiting... Then the next load is significantly easier (it will load and work in about 2 tries) however, if I forget to resize window and leave it at full screen then it takes 8 or 9 yes for it to load. It's very odd. I do have additional displays (4k tv, dual curved 1440p displays, and index) I don't think ive tried eliminating displays recently. I'll try it when I get home and report if it changes anything.

Kuxer, 2021-12-22:
I just can run Cura after I killed and start explorer.exe. I don't understand, why did it help, but now cura work. Yesterday I rebooted my computer about 15 times)

Ghostkeeper, 2021-12-30:
Could it be that you're running out of GPU RAM?

RyzenSavior, 2021-12-30:
I doubt it. I have rx 6800xt with 16gb. 32gb cl 14 3600mhz sys ram. Dark
hero mobo and Ryzen 9 5950x. On custom water loop. I keep my system pretty
up to date. And have suffered this through many different versions of this
build. Although all have been AMD builds. I did get the new beta version of
cura and it is better. Not it ALWAYS works on the third opening. Although
when it does it logs me out and gives me a cannot communicate with cura
server popup. This further leads to my suspicion that it some kind of auth
server request that's failing to go through and that actually causes the
crash. I use Kaspersky av and of course I've white listed cura and provided
full firewall access...

GregValiant, 2024-11-16:
Is this still a problem in current versions of Cura (5.8.0 and up)?  Can this be closed?

closed 2024-12-09 by github-actions[bot]