Ultimaker/Cura #10932: More problems with 4.12 because of its new conflicts with Klipper Firmware.

RyushoYosei, 2021-11-26:
I explained thin in my old report, but cura 4.12 causes prints to fail or have significant issues still with Klipper firmware, which reads the same marlin gcode, but because it has built in features for controlling acceleration, and preventing ringing, Cura's built in ones now cause issues with its acceleration and flow control, making prints either fail from skipped steps that the printed w ouldn't have in 4.11, or under extrusion. I have tried the same print, same slicing in both versions, and the 4.12 one always fails, the 4.11 one always succeeds, and there is no difference in the print settings at all. 

For example I have a print I have printed successfully on my Klipper firmware CR 6 SE with the acceleration set to 5000, However when I do the exact same settings, s liced with 4.12, the printer skips steps and the print fails, unless I lower the accceleration in the firmware all the way down to  2000, and even then it still has under extrusion issues caused by the gcode 4.12 inserts to help prevent ringing for normal Marlin printers.

RyushoYosei, 2021-11-26:
The old issue that is closed, is [https://github.com/Ultimaker/Cura/issues/10793](url) But still, it is an issue that is coming up because of the changes in 4.12, as all previous versions were working fine with it.

RyushoYosei, 2021-11-29:
[Correct link to the old issue @_@]( https://github.com/Ultimaker/Cura/issues/10793)

fvrmr, 2021-11-29:
Hi @RyushoYosei thank you for your report. 
I will bring this up with the team. Keep you posted!

fvrmr, 2021-11-30:
I have discussed this with the team. And we decided to reject this, we can't test this ourselves since our printers are not using Klipper firmware. A pull request is always welcome. I hope you understand.

closed 2021-11-30 by fvrmr