Ultimaker/Cura #10935: Cura crashes when opening a STL file.

Ressless4, 2021-11-27:
### Application Version

4.9.0 - 4.12.1

### Platform

Windows 10

### Printer

Anyubic Mega S

### Reproduction steps

Opened STL from menu
Drag an Drop STL into program
Used that method for Cura 4.9.0, 4.10.0, 4.12.1 and 4.12.0

### Actual results

Hi I have "almost" the same issue / problem with the crash and tried the fix mentioned for Simulatorview but it didnt work.

My old  4.9.V Version worked around March but now it doesnt with 4.12. and 4.9. as well. 

 What I did was updating my graphics driver.

Can someone help me please?

Also this happens as well with all never versions(already tested).

### Expected results

it should open and render the STL File

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[image]

`Traceback (most recent call last):
  File "C:\Users\ressl\AppData\Roaming\cura\4.9\plugins\CustomJobPrefix\CustomJobPrefix\PrintInformationPatches.py", line 120, in _triggerJobNameUpdate
    self._updateJobName()
  File "C:\Users\ressl\AppData\Roaming\cura\4.9\plugins\CustomJobPrefix\CustomJobPrefix\PrintInformationPatches.py", line 129, in _updateJobName
    base_name = self._print_information._stripAccents(self._print_information._base_name)
AttributeError: 'PrintInformation' object has no attribute '_stripAccents'
`

[cura.log](https://github.com/Ultimaker/Cura/files/7611341/cura.log)

fieldOfView, 2021-11-27:
This issue is fixed by updating the "Custom Printjob Naming" plugin from the Marketplace. That update has been available since before Cura 4.9 was released.

Ressless4, 2021-11-27:
Thank that solved it!

closed 2021-11-27 by Ressless4