Ultimaker/Cura #10936: Prepare and preview windows unresponsive.

GrooverFromHolland, 2021-11-27:
### Application Version

Arachne_Engine_Beta_2

### Platform

Windows 11

### Printer

Ender 5 plus

### Reproduction steps

First scenario: Open Cura, select file from menu.
Second scenario: In file explorer, select file, open with Cura. 

### Actual results

First scenario: Prepare window absolute unusable, lag, no response and waiting forever. Slicing is OK. 
      Preview window is also unusable, lag, no response and waiting forever.

Second scenario: No problems 

### Expected results

I have the log files for both scenarios as Cura1.log and Cura2.log

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura1.log](https://github.com/Ultimaker/Cura/files/7611725/cura1.log)
[cura2.log](https://github.com/Ultimaker/Cura/files/7611727/cura2.log)

fvrmr, 2021-11-30:
Hi @GrooverFromHolland thank you for your report.
I can't test this myself since I am on Win10 but I will check with one of my colleagues.
Keep you posted.

fvrmr, 2021-12-07:
@GrooverFromHolland do you also see behaviour like described here: https://github.com/Ultimaker/Cura/discussions/10931#discussioncomment-1748963 in the preview?

GrooverFromHolland, 2021-12-07:
No, everything is just dead slow. Mouse clicking, dragging and keyboard input takes 3 to 4 seconds to respond. Preview animation is almost standing still when rounding corners.

fvrmr, 2021-12-14:
Can you install our plugin: https://marketplace.ultimaker.com/app/cura/plugins/UltimakerPackages/ExtensiveSupportLogging this will generate hopefully more information for us.

GrooverFromHolland, 2021-12-15:
I installed the plugin, not sure how to use it.
This is what I did:
I opened Cura, selected STL. Selected my object on the build plate and moved it to x 0 with keyboard. All keyboard input, like delete, select, backspace etc, takes about 5 seconds each, sometimes longer(window title shows not responding). 
Mouse-click on icons also takes about 5 seconds or more, but selecting object on the build plate is immediate.
Also selecting object-face to lay flat on build plate and the rendering is immediate.

 Slicing takes about 2 seconds.
 In preview, I Used the sliders, to show show layer 1 and fast-forward, sliders are unresponsive fore some time. Preview is very slow.
 Saved the remote-support file and closed Cura.
[remote-support_21-12-15-14_11_27.zip](https://github.com/Ultimaker/Cura/files/7719976/remote-support_21-12-15-14_11_27.zip)

sazhnikoff, 2022-01-11:
I have the same problem on Windows 10 (latest official build with all updates, eng language)
Maybe the problem is in the hardware or driver version?

Several times Сura was able to show the layers and there were clearly graphic artifacts .... Other versions work fine for me, including 4.13-beta and arachne beta 1

Ghostkeeper, 2022-01-18:
The remote support archive we got there only contains a log file. Sorry, we should've provided better instructions there.

Since this issue is about performance and we can't reproduce it ourselves, what we'd need to know is what Cura is busy with on your computer. The ExtensiveSupportLogging plug-in can measure that, but only does that when commanded. So these are the steps we'd like one of you to take:

1. Install the plug-in (if you haven't already).
2. In Cura, click on Extensions -> Remote Support -> Start CPU profiler from the application menu.
3. Do the thing that is slow, in your case opening a file from the menu.
4. Click on Extensions -> Remote Support -> Stop CPU profiler.
5. Click on Extensions -> Remote Support -> Save remote support archive.
6. Post the resulting archive here.

If the issue only happens with opening files via the file dialogue but not with files dragged into the Cura window, I'd suspect that something in the file dialogue is causing the slowdown. Perhaps Qt/Windows is doing something in the background with a network drive or something?

GrooverFromHolland, 2022-01-18:
I want to add that this problem is on all Cura versions since version 4.8.
[remote-support_22-01-18-13_32_11.zip](https://github.com/Ultimaker/Cura/files/7889158/remote-support_22-01-18-13_32_11.zip)

Ghostkeeper, 2022-01-25:
Thanks for that support archive! Here are the top few functions (sorted by "own time") it's spending time on:

[image]

I'm not sure if I see anything special here. The `_setUniformValueDirect` call being up there is normal, part of the rendering process (for me it's normally about 10%). It's slightly weird that that `inner` function of the typing module takes up that much time while it should be a no-op, but in the grand scheme of things it's not really something you would expect Cura to freeze with for minutes. No single function took up more than 687ms.

It spent 4343ms on rendering in total. This is about the same as the claimed delay after opening a file from the file dialogue (3-4 seconds at every click) except that it did 4875 renders in that time, so each render is under 1ms which is very fast.

So it's something that doesn't show up in here, meaning it's not something in Python code that is causing the slow down. Unless it's opening a 3MF file (for which we have [a C++ module](https://github.com/Ultimaker/libSavitar)) this means that it's not slowing down in our own code, but in one of our dependencies. Given that it happens when the file dialogue is opened, I would guess that it's something in that file dialogue. Perhaps a thumbnailer desktop integration is still generating thumbnails in the background, or maybe something is checking the availability of network drives or something.

One thing you could try is disconnecting from the internet to see if it still happens then. We've seen before that applications like Google Drive or Dropbox can influence Cura (in their case, cause Cura to crash) merely by Windows wanting to display their options in the file menu it creates by itself.

GrooverFromHolland, 2022-02-15:
More information:
Starting Cura from STL-file makes the problem go away, but only once! If I close Cura and do it again, the problem is back. Only restarting Windows makes it go again, but only for one time.
It also looks related to   #11467

fvrmr, 2022-02-25:
@GrooverFromHolland on the other issue we also don't know what is causing this. We have to wait for more reproduction steps / information so we can look into this for this issue as well.

GrooverFromHolland, 2022-02-25:
I Uninstalled Ultimaker Cura Arachne_engine_beta and Cura 4.13.1 . I removed all remaining files and folders with Ultimaker or Cura in the name from C:\Program Files, C:\Users\"ME"\AppData and C:\ProgramData\Microsoft\Windows\Start Menu\Programs.

After reinstalling Cura Arachne_engine_beta , all works again as it should.
It seems to me that the uninstaller doesn't do a very good job, leaving a lot behind from previous installs.

Ghostkeeper, 2022-03-12:
The uninstaller is expected to leave your settings where they are unless you click "yes" in the pop-up that appears during the installer. When clicking "yes", it's supposed to remove everything except the Qt shader cache, which are shared with other applications. But if you didn't instruct it to erase the settings, it will indeed not remove everything.

GrooverFromHolland, 2022-03-12:
So the  Qt shader cache is likely the cause of all the troubles I and many others had the last half year.

Ghostkeeper, 2022-04-02:
Maybe. You can test that by deleting the Qt shader cache as well. I'd expect it to be somewhere in your local AppData. I've deleted that in the past as well.

It's not something Cura should touch, but it's really quite safe to do. If other Qt applications are still running it might give rendering errors with those.

GrooverFromHolland, 2022-04-02:
I can't test that. By removing all traces of files from previous installations and re-installing Cura my problems were solved. Maybe  others with the same problems can try.

closed 2024-12-09 by github-actions[bot]