Ultimaker/Cura #10937: Cura Crash while opening marketplace

UtzLe, 2021-11-27:
### Application Version

4.12.1

### Platform

Mac OS X, Catalina 10.15.7

### Printer

Creality Ender 3

### Reproduction steps

I wanted to install a plugin from the marketplace. So I clicked on "Marketplace" and waited a few seconds

### Actual results

Cura crashed. So I had opened it again and tried again to open the Marketplace. It crashed again.

### Expected results

I expected that the marketplace open correctly, so I can install an plugin.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7611919/cura.log)

fvrmr, 2021-11-30:
Hi @UtzLe thank you for your report.
We can't reproduce your issue and your logs are showing a hard crash. So it is hard for us to know why it is crashing.
I will leave this issue open and we will have to wait if somebody else runs into this issue and hope we can reproduce it.

fieldOfView, 2021-11-30:
Did Cura actually crash (as in: the application window closed) or did it become unresponsive (as in: the application no longer responded to input, and you force-quit it after a short while)? If it is the latter, could you try again and give the Marketplace window more time to open? 

On my windows development machine it takes up to a full minute of the Marketplace being unresponsive (seemingly crashed) before the Marketplace window is populated. The Cura team is aware of that, and are rewriting the Marketplace to fix this issue - among others.

UtzLe, 2021-12-03:
> Did Cura actually crash (as in: the application window closed) or did it become unresponsive (as in: the application no longer responded to input, and you force-quit it after a short while)? If it [...]

UtzLe, 2021-12-03:
Cura crashed and the application closed almost immediately after opening the Marketplace

llalexander00, 2021-12-05:
Same experience here. 

Cura crashed and the application closed almost immediately after opening the Marketplace

fvrmr, 2021-12-07:
It could be related to the MacOS security update. https://github.com/Ultimaker/Cura/issues/10974
But we are not able to reproduce this issue.

closed 2024-12-09 by github-actions[bot]