Ultimaker/Cura #10939: halving the layer height setting for "modify settings for overlaps"

TwomD, 2021-11-27:
### Is your feature request related to a problem?

if i have a model that i want to print at a high Layer height ( Eiter for speed or esthetics ) but there are parts that require small layer heights ( shallow curves ) to avoid big stair stepping's.

### Describe the solution you'd like

would it be possible to add a function that halves the layer height vor overlapping objects ( using Support blockers -> modify settings for overlaps -> something like "1/2 layer height" or even 1/3  )

basically something that already exists for infill: "Infill Layer Thickness" but instead of doubling the layer height, halving it. ( in a specified region ) 

something like this: 
[image]

the layer height for the part would be 0.28 but the wall layer height of the selected region would be half of that 0.14
that way the visibel layer height on the outside is continuous and the inside round part wouldn't suffer from the topographical effect so much.

- the basic version would be to halve all the layer height in the specified region and a more refined approach would be to only affect certain line types ( as in walls ). that way it would be easier to "aim" ;)
- or maybe a function where you can select a surface ( but that sound way harder to programm... )
- a different approach could be to define a wall angle at wich it halves the wall layer height for walls ( now that i think about it, that would be the most user friendly option i guess ) 

### Describe alternatives you've considered

right now i only see 3 options:
- accept the stair stepping effect :(
- reduce the layer height for the whole print -> takes a lot longer + for some parts i want thicker layer lines cause of esthetics
- using variable layer heights -> that leaves uneven layer heights on parts that are not desired 

### Affected users and/or printers

will effect everyone that wants to print fast with bit layer lines but there are small parts of the print that require finer layer lines.

### Additional information & file uploads

didn't find any similar request ( maybe i didn't search hard enough... im new to GitHub ;) )
and im not a coder but since something similar already exists ( infill layer thickness ) i think it wouldn't be that hard to adapt it ( correct me if im wrong ;) )

fvrmr, 2021-12-02:
Hi @TwomD thank you for your feature request.
We already have adaptive layer heights under experimental. But that is for the whole model. 
But I will bring up with the team. Keep you posted!

fvrmr, 2021-12-03:
I have discussed this with the team and we all agree that this would be a really nice feature. 
However it is really complex to implement and it would take a lot of time. Because we also need to think about how it will work with multiple models. And with multiple models the changes are that you will bump into another model with the nozzle. 
So that's why we decided to defer this issue. I hope you understand!

TwomD, 2021-12-03:
thanks for the info :)

closed 2023-07-13 by github-actions[bot]