Ultimaker/Cura #10940: Cura 4.12.1 Mac crash on start

scmugan, 2021-11-27:
### Application Version

4.12.1

### Platform

Mac OS X 10.15.7 

### Printer

Ender 3 Pro

### Reproduction steps

Opened Cura

### Actual results

It goes through the start up screens and then goes away. After a bit the Mac app OS crash screen pops up

### Expected results

Cura should load up and let me work

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7612316/cura.log)

Abacab7711, 2021-11-29:
I have the same issue. Attached is the log files.
[CuraCrashlog.pdf](https://github.com/Ultimaker/Cura/files/7618960/CuraCrashlog.pdf)

scmugan, 2021-11-30:
I fixed this by going to Macintosh SSD > Users > _username_ > cura and changing the name of the 4.12 folder to 4.12-old and then Cura was able to fire up

fvrmr, 2021-11-30:
Hi @scmugan thank you for your report. 
Good to know that Cura won't crash for you. 

@Abacab7711 could you try and see if that works for you to?

Abacab7711, 2021-11-30:
Did not work - I even made a new user account and the same happened. Could this have anything to do with Python?

fvrmr, 2021-12-07:
I think it is related to the security update from MacOS. But we haven't been able to reproduce this ourselves. See: https://github.com/Ultimaker/Cura/issues/10974

Ghostkeeper, 2021-12-08:
> I fixed this by going to Macintosh SSD > Users > username > cura and changing the name of the 4.12 folder to 4.12-old and then Cura was able to fire up

That means that the crash is caused by something in the configuration folder then. I'd be interested to see that 4.12-old folder if you still have it.

scmugan, 2021-12-08:
Here it is
[4.12-old.zip](https://github.com/Ultimaker/Cura/files/7677402/4.12-old.zip)

Ghostkeeper, 2021-12-09:
Hmm, sadly that doesn't make it reproduce for me. It properly loads up Cura, with your Ender 3 Pro added and infill density set to 15%, etc. Only I see that you were logged in but it can't log me into the account because the log-in secret is not stored in the config folder (for security reasons; it's in the OS keychain to store it securely).

Unless this has something to do with the account, it seems that the issue is not just specific to the configuration, but also specific to something in the computer's environment (e.g. MacOS).

closed 2024-12-09 by github-actions[bot]