Ultimaker/Cura #10941: Cura shuts down attempting to open a file

bruster555, 2021-11-27:
### Application Version

4.12.1

### Platform

Windows 8.1pro

### Printer

JGAurora A5

### Reproduction steps

Updated to this version, now when i try to open a file it shuts cura down and i have to restart it. Even after a restart, when i try to select a folder to open, it shuts down??
Very frustrating. May have to go back to 4.7 which worked with out an issue.

### Actual results

See above

### Expected results

Should be able to open files to print with out shutting down the program.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Non at this time, could not locate log files

fieldOfView, 2021-11-28:
If you have Chitubox installed, please read this article for a fix: https://support.ultimaker.com/hc/en-us/articles/4402996140946-Ultimaker-Cura-crashes-when-opening-any-file

If not, please check if there is an update available for your plugins in the Marketplace.

AlexHulstaert, 2021-11-28:
Try turning off preview file in the open file dialog (ALT P)
I have this problem too. It seem to be the file preview affects Cura with some STL files.

Cheers Alex

closed 2021-12-13 by no-response[bot]