Ultimaker/Cura #10942: Wireframe Printing Can't slice with Knot strategy.

RyushoYosei, 2021-11-28:
### Application Version

4.11

### Platform

Windows 10

### Printer

Creality Cr-6 SE

### Reproduction steps

Slice a model with wire printing, set to strategy Knot.
Let it automatically attempt to slice.

### Actual results

Failed to slice

### Expected results

Sucessfull slicing

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[CCR6SE_DragonStatue.zip](https://github.com/Ultimaker/Cura/files/7612684/CCR6SE_DragonStatue.zip)

fvrmr, 2021-11-29:
Hi @RyushoYosei thank you for your report. 
I will bring this up for discussion with the team. Keep you posted.

fvrmr, 2021-12-03:
I have discussed it with the team and we decided to defer this issue. 
Wire printing is in experimental and it is not a priority for us to look into this. I hope you understand.

closed 2023-06-09 by github-actions[bot]